bigtreetech/BIGTREETECH-SKR-mini-E3 #501: Hello i installed firmware-bltouch-for-z-homing.bin and now i have a blue screen

Sophia7979, 2021-01-31:
### Description
i have a ender 3 with the skr mini e3 v2.0 board installed and also a BLtouch 3.1 after installing the bltouch i read some instructions online for updating the firmware for the BLtouch i downloaded firmware-bltouch-for-z-homing.bin renamed it firmware.bin put it on my sd card and started the printer with said sd card in the printer well it just gave me a black screen after a few min went by i shut the printer off assuming that maybe it just needed to be rebooted and now when i start the printer with or without the sd card the screen just stays black

**Expected behavior**
i expected for the firmware to be installed and a BLtouch option be added to my settings so that i could use my BLtouch

**Actual behavior**
blue/black screen of death

### Additional Information
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V1.2/firmware-bltouch-for-z-homing.bin?fbclid=IwAR1QLclc6TyKkLbxDXlJGLvVXcjRY6Xc4f6cB7tDFvSZWXwZHLF1uDJdxwY

thats a link to the file i downloaded 

https://www.instructables.com/Install-BL-Touch-on-Ender-3-With-SKR-Board/

those are the instructions i was following

nonflammable, 2021-01-31:
it looks like you have used the firmware version for v1.2 board, but you have v2.0 board.
see if you can update with this file:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-bltouch-for-z-homing.bin
(remember to rename)

Sophia7979, 2021-02-06:
> it looks like you have used the firmware version for v1.2 board, but you have v2.0 board.
> see if you can update with this file:
> https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-bltouch-for-z-homing.bin
> (remember to rename)

Thank you so much i guess im just blind lol