bigtreetech/BIGTREETECH-SKR-mini-E3 #504: BTT SKR Mini E3 V 2.0 Thermal Runaway Error every time it heats up over 100 Celcius

wprojects, 2021-02-02:
What I am working with:
**OS: Mac OS X** 
-- Catalina
**Board: BTT SKR Mini E3 V 2.0**
--Marlin 2.0
**Ender 3**
--The stock screen isn't working with my board, I have tried the ports and no luck. The screen turns blue and blank. I could get it to work with the original board by adding a second lcd cable to the port, but this one only has 1 LCD Port
**Software**
--Cura
--Pronterface

I'm running a USB from my Mac to my board, and every time I try Pronterface or Cura it starts to heat up but then the thermal runaway triggers anywhere from 94 Celcius to 120 Celcius (Typically between 110-120 Celcius). I have never seen it make it over 120 without it triggering the thermal runaway. I am pretty stumped on what to do from here, any help would be greatly appreciated!

**Note:**
Originally to get Cura to even recognize the board via usb I had to edit a file, but everything is working now other than the thermal runaway triggering which is happening on pronterface as well.

**What I have tried:**
1. Unplugging usb and restarting the device
2. Restarting the printer software in addition to step 1
3. Loosen the thermistor screw. I saw this was a common issue but I actually replaced the thermistor with a brand new one, so I highly doubt this is the issue
4. Pressing the reset button on the board
5. Typing m303 and m501 m502 commands in pronterface for a PID auto tune (Got the same results)
6. Checked the cords to make sure everything is in there tightly
7. Contemplated throwing my printer out the window as a free donation for a sweet paper weight

**What I would like to solve**
1. Fix the thermal runaway trigger
2. Solve the blank LCD Issue (This one would be more of a bonus! My top priority is getting the printer functional)

I cannot find much information regarding this issue. I saw that the issue might be related to the usb cable not having enough power but this has worked in the past on my old heater cartridge before I messed up the thermistor on it. They said it was fixed by unplugging or resetting I think which did not work for me. Also I think theirs might have been beeping from the start and mine is heating up to over 100 celcius typically. 

Any help would be greatly appreciated!
W

hawkuil, 2021-02-10:
Hi there got almost the same problem in my ender 5 pro. Only difference is that that it heats up fine if i put the temps in manual but with the heat up profiles PLA and petg i get the error.

hitokiri8x, 2021-02-10:
Sorry but saying 

> "I saw that the issue might be related to the usb cable not having enough power but this has worked in the past on my old heater cartridge before I messed up the thermistor on it"

Are you meaning that you are powering all of it from your pc? and not from the 24V power supply?
Also have you followed
https://3dprinting.stackexchange.com/questions/7610/getting-e1-thermal-runaway-errors-even-after-replacing-the-heating-element-on-t  ?

mikey369, 2021-04-08:
Ender 3 Pro w/Swiss Hot End, BLTouch. Replaced Creality 32 bit board with BIGTREETECH Direct SKR Mini E3 V2.0 32Bit Control Board + TFT35 E3 V3.0 Integrated with TMC2209 UART VS TMC2208 for Ender 3 Pro/5 3D Printers. No issues w/Creality 32 bit board prior to SKR replacement.

I have a similar bed overheating issue. The LCD and the stock Creality displays work fine on the SKR boards. However, my bed starts heating immediately after I turn on power until thermal runaway kicks in (or I turn it off). From both display panels I sent commands to set the bed to 60C and the info screen shows the command took (has 60C on the top) but temp keeps climbing (on the bottom) till thermal runaway kicks in - also around 110C. 

PlatformIO fails installing in VSCode and I've not found all the places I need to remove VSCode from so I can get a clean new install to recompile. Bottom line is I don't know if this is firmware or a wiring error. Any guidance would be greatly appreciated. I realize you have no control over VSCode/PlatformIO issues but maybe there's an alternative I'm not aware of being a 3D printer noobie...