bigtreetech/BIGTREETECH-SKR-mini-E3 #508: [BUG] (short description)

rennurb3000, 2021-02-04:
### Description
z - homing with bltouch does not work ( ender 3 (standard) and skr mini 1.2)

### Steps to reproduce

1. Download the firmware V1.2/firmware-bltouch-for-z-homing.bin or build it yourself with appropiate configuration
2. put it on the SD card and rename it to FIRMWARE.CUR
3. Start homing (All axis or Z axis , does not matter)

**Expected behavior**
Z axis should move up a bit then probe would deploy and move down till the probe should recognize touchdown

**Actual behavior**
Probe is not deployed , zaxis will hit the ground
(like normal z homing. i also tried hooking up the z stop like normal and it does normal z homing)

YoAdriaaannn, 2021-02-21:
Does the probe deploy when you try to manually deploy it through the panel? If not you probably have it wired up wrong.

rennurb3000, 2021-02-21:
the probe can be deployed in manual mode. however while homing it just does not use the sensor. it also looks like i am not the only one with the problem as there are multiple issues with bltouch and homing now . bltouch definitly works, since i also tried the **other** configuration, where the sensor is connected to the probe, not the z stop and i was able to do abl . There might simply be a bug in the firmware. ;)

Dlittle2000, 2021-04-01:
Did you ever find a fix. Also they just updated firmware 10 hr ago.