bigtreetech/BIGTREETECH-SKR-mini-E3 #509: Error compiling when trying to set up Neopixel

jorgedemi, 2021-02-06:
I've been trying for quite a long time to get my Neopixels to work, I've followed many github and reddit posts about this, but I'm stuck in the error I show on the picture. I replace at line 221 of platformio.ini " Adafruit NeoPixel@1.5.0 " with "[https://github.com/CommandoreBombardiero/Adafruit_NeoPixel](https://github.com/CommandoreBombardiero/Adafruit_NeoPixel)" as shown in many tutorials, but I get this error when I compile. I've tried adding the path to the non-blocked list of Avast as it was suggested in another post but the error keep occuring. Does anybody know how to fix this?

[image]

P.S. I've been following this guide: [https://gist.github.com/877dev/2f8cd65118757afb24b7edd6523d09f9](https://gist.github.com/877dev/2f8cd65118757afb24b7edd6523d09f9)

NordicTroll, 2021-02-18:
I am having the exact same problem. 
You're not alone.

jorgedemi, 2021-02-19:
> I am having the exact same problem.
> You're not alone.

I finally managed to do it thanks to this tutorial:
https://www.youtube.com/watch?v=ianwpYJVyUs&ab_channel=3DPrintscape

It's great and it explains this step by step.

closed 2021-02-19 by jorgedemi