bigtreetech/BIGTREETECH-SKR-mini-E3 #510: [BUG] Bigtreetech SKR Mini E3 V2.0 - Broken nozzle cooling fan, twice.

Piket564, 2021-02-13:
I have successfully installed the card on my Ender 3 Pro, I started to print correctly configuring language, bltouch and all the changes that interested me, after about 18 hours of printing (discontinuous, I made an average of the prints) the cooling fan of the nozzle burned out.

Inserting from the screen (or even with octoprint) the fan at 70%, it starts for a few seconds at reduced speed, but then stops immediately afterwards.

I already have 2 fans burned out in a short time, I still have a third fan, but I would like to avoid burning it.

I attach as required, Configuration.h and Configuration_ADV.h

Here the files: [conf.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5974803/conf.zip)

DarkSnart, 2021-02-13:
Picket564-   Are you using Brushless fans by chance? Maybe this thread will help you. 

**[[BUG] Straight PWM destroys brushless DC fan motors. #439]**

radek8, 2021-02-13:
I was bothered by the fan noise during PWM control.
The 1uF / 60V electrolytic capacitor connected in parallel to the fan helped the most. He smoothed the edges of the PWM signal and the fan fell silent.
It could also protect the sensitive fan from destruction.
[image]

Piket564, 2021-02-21:
> @Picket564- Are you using Brushless fans by chance? Maybe this thread will help you.
> 
> **[[BUG] Straight PWM destroys brushless DC fan motors. #439]**
> **Also Issue #238 comment by ICM69 may apply**

I'm using DC fans bought from amazon.

I will try the @radek8 solutions and we will see!
Thanks!