bigtreetech/BIGTREETECH-SKR-mini-E3 #513: [BUG] printer`s moves to the end at X ,Y axis after each layer change

aasjoa, 2021-02-17:
after I switched from the orginal crealty mainboard to new mainboard BIGTREETECH® SKR MINI E3 V2.0 on both Ender 3 pro, the printer`s moves to the end at X axis, end the table  also moves out after each layer change, throughout the entire print`s 
that are not right?

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the BigTreeTech Facebook group: https://www.facebook.com/groups/505736576548648 or BigTreeTech Subreddit: https://www.reddit.com/r/BIGTREETECH/.
-->
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

sarvenn, 2021-02-24:
Does this happen at a certain gcode or everything you print happens this issue?
It looks like your gcode has a photo capture scripts at every layer. Could you check your slicer settings and gcode lines for this?

aasjoa, 2021-02-24:
Hi. this happens on everything I print. Yes i will check.

aasjoa, 2021-03-14:
I have done this and the same thing happens again, I am considering
putting back the original motherboard on both prints.

joachim

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>