bigtreetech/BIGTREETECH-SKR-mini-E3 #517: Part cooling fan not spinning

arc9531, 2021-02-18:
I have been searching and searching and searching and find no support for the problem I am having which is the parts fan (filament fan) does not spin up at all, whether I turn it to 100% on the controller or to 10%, no spin at all. I have the skr mini v2.0. This is really pissing me off. I wired the v2.0 board just as it was on the base ender 3 pro board. Please help me!

sarvenn, 2021-02-19:
I had the same issue with the latest marlin bugfix2 installed to my skr mini e3 v2 and tft35 with the latest firmware.
Issue only occured in TFT mode both in printing or at main status screen. When I switched to marlin mode I could manage to run the cooling fan properly in status screen or in printing.
Issue doesn't happens all the time but intermittenly happens. I changed nothing but now it runs ok. I thought processor might be overloaded etc. I don't know.

DarkSnart, 2021-02-23:
@arc9531 @sarvenn This seems to be an ongoing issue with this board and it's conformity to some of Marlin's prerequisites, OR vise-versa. My problem was FAN0 was always on, and when I increased the speed, the fan rpm would actually decrease, and nothing would turn it off. After a week of searching I found my case to be a burned mosfet, and followed a work-around which required editing the pins file and re-assigning FAN0 to FAN1, and subsequently using FAN1 as my parts cooling. [https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/277](url) I then connected my board fans and extruder fan to 24v+ const on the board.  As for those who do not have a known mosfet issue, it appears that it still may be... the mosfet. I suggest this only because of the lengths I've traveled and things I've seen concerning this board and this circuit in particular. There was a post about turning off the speaker setting in Marlin to enable FAN0 to run appropriately. WTFE. I too would like to find out the root cause because I have three of these boards; one which works as expected, one with burned mosfet and one still new on the shelf which I plan to use in another application soon. Some are noting this issue right out of the box which further supports my theory of weak mosfet from the factory.

sarvenn, 2021-02-24:
Hmm. I had encountered this issue after uploading a firmware which has speaker as enabled. By the way I just had this issue after uploading the firmware and doing my mesh bed leveling, changing some other parameters for my machine etc. After that I never see that issue again almost for 20 days.
But I will undefine speaker at my next firmware update. Thanks for the tip.