bigtreetech/BIGTREETECH-SKR-mini-E3 #520: SKR Mini v2.0, Ender 3 Pro: Z-Axis does not go down

cg089, 2021-02-28:
Hi,

I swapped my 4.2.2 mainboard which came with the Ender 3 Pro and the original BLTouch 3.0 (5-pin) with the SKR Mini v2.0.

Mainboard boots up fine. I flashed the firmware-bltouch, but "Auto Home" moves the Z-Axis only upwaords, not downwards, so  there is no homing at all.

Manual movement of the Z-Axis only goes upwards as well, since I can't enter negative numbers.

Any ideas on what went wrong?

revjkramer, 2021-03-06:
I have the same issue.... have you been able to resolve this?

cg089, 2021-03-06:
It seems it only works with the BLTouch AND the Z-switch. After I reinstalled the Z-switch, it worked somehow.

If you want to get rid of the Z-switch, I guess you would need to have a different BLTouch sensor which has three pins connected to the 5-pin-connector and 2 pins to the Z-switch-connector.

Or maybe build your own firmware which swaps the PC14 and the PC2 pins?

revjkramer, 2021-03-06:
Ok here is what I did.

1. Disconnected the white and black wire from the portion of the board that was connected next to the other wires for BLTouch. I connected it to the z end stop.
2. Flashed the firmware for Z Homing and now all is well.

I couldn't reconnect the z stop because I just switched to the BL Touch from an EZAble knock off which wouldn't level right for printing on glass. I had been using that for about a year and I have no idea what I did with the switch.

BrayPagan, 2021-04-16:
same problem, getting tired of having to figure out why new stuff doesnt work.  You would think the company would try it out first .