bigtreetech/BIGTREETECH-SKR-mini-E3 #521: [BUG] (short description)

masterdot, 2021-02-28:
Hey there, 

BTT SKR Mini 3V2 on a Ender 5 Plus with Marlin Tag 2.0.7.2

I wanted to enable automatic Z leveling with marlin, but I cant address the second z axis.
It looks to me that the second z stepper is not enabled via config. 
Maybe you can check that out

`
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34:0,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:1410:8: error: #error "No E stepper plug left for Z2!"
       #error "No E stepper plug left for Z2!"
        ^~~~~
`

hitokiri8x, 2021-03-03:
That will not be possibile. the second stepper is not configured because there is no drive for it! 

Basically there is only one driver with its output splitted in 2, so every stepper will receive the same command.
This means that you can't command individually a z motor, nullifying the g34 command.

masterdot, 2021-03-03:
Thanks for the info, did find it out some hrs ago myself :) currently thinking about switching to a skr 1.4 turbo to have this option.

closed 2021-03-03 by masterdot