bigtreetech/BIGTREETECH-SKR-mini-E3 #522: [BUG] Serial Number is 0

sp00ls, 2021-03-03:
### Description

The serial number of the SKR E3 Mini V1.2 board is reported as 0. This makes it impossible to write a udev rule to guarantee the device location when plugging multiple boards into one computer.

### Steps to reproduce

1. Plug in the device
2. Check the system log to see the device serial number `dmesg`

**Expected behavior**
The serialNumber to be unique

**Actual behavior**
The serialNumber is always 0:
[ 1214.999619] usb 1-1.3: new full-speed USB device number 14 using xhci_hcd
[ 1215.136145] usb 1-1.3: New USB device found, idVendor=1eaf, idProduct=0004, bcdDevice= 2.00
[ 1215.136167] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1215.136185] usb 1-1.3: Product: Maple
[ 1215.136203] usb 1-1.3: Manufacturer: LeafLabs
[ 1215.146807] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device

radek8, 2021-03-06:
Have you tried to allow in Marlin?

#define MACHINE_UUID

sp00ls, 2021-03-06:
> Have you tried to allow in Marlin?
> 
> #define MACHINE_UUID

Yes I have, there was no change in the udevadm output