bigtreetech/BIGTREETECH-SKR-mini-E3 #523: [BUG] (USB connection)

agio26, 2021-03-03:
### Description

### Steps to reproduce

1. Connect the printer to USB cable
2. Connect the USB cable to the computer

**Expected behavior**
Connect the USB cable without any issue and possibility to use cura or proterface

**Actual behavior**
A sound like an alarm is triggered. On the computer, I don't have CURA, pronterface, or any other software that can connect to the printer

### Additional Information
Hardware: SKT mini E3 V2 + TFT35 + BLTouch
Firmware: Marlin 2.07.2 (I tried official firmware-bltouch.bin and firmware-bltouch-for-z-homing.bin)
Cable: original cable and others that I had
TFT35 use 2 cables (serial and display)

trengtor, 2021-03-03:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/518

KLOSTERS, 2021-03-03:
Maple Driver cannot be a solution: It is last updated 2016. Windows has reliable CDC drivers nowadays.
My SKR-Mini-E3 V2.0 with the original software worked just fine, until i compiled and and flashed firmware dated by 04.02.2021.
No Serial Mode for the BigTreetech TFT24 Display, no USB Connection.

Current config:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 3

KLOSTERS, 2021-03-08:
Does your Board work otherwise?