bigtreetech/BIGTREETECH-SKR-mini-E3 #524: Random pauses in LCD Simulator mode

dmarnel, 2021-03-05:
### Description

### Additional Information

Marlin 2.0

radek8, 2021-03-05:
Do you use the SD card slot on the TFT or SKR to print in Marlin mode?

dmarnel, 2021-03-05:
Actually, neither, I print over WiFi using Cura via Octoprint.

radek8, 2021-03-05:
What is the connection structure?

dmarnel, 2021-03-05:
Not sure what you mean by that. I use an API key provided by Octoprint over a local private network, if that helps.

radek8, 2021-03-06:
You have a SKR, display board TFT, WIFI, Octoprint.
I ask what is connected to which connector.

You have quite a specific configuration. If it works for you when switching to touch mode, then use it :-)

dmarnel, 2021-03-06:
And I do use it, most of the time. However, the M600 (filament change) does not work correctly in touch mode, which is another issue that needs to be looked at. It doesn't present any way to resume after the filament is unloaded.

SamppaD78, 2021-06-04:
M600 (filament change) or filament detection run out does not work correctly in touch mode with skr mini e3 2.0 is exactly the same issue that i have been having. No solution my friend. I have try everything,updating marlin,btt tft 50 screen,compiling marlin different versions. With different versions of marlin you get different issues. It looks like the screen is not communicating correctly with main board. G codes that screen send are accepted on the screen but  ignored by main board!?

dmarnel, 2021-06-04:
I was able to get the M600 filament change and some other issues working by using the bug fix version of Marlin (bugfix-2.0.x). Not sure why this hasn’t been made an official release yet.

SamppaD78, 2021-06-04:
Hi.Great news.I didn't know that,as i always using the latest Marlin release. I have to try it. Thanks for tip . By the way this issue regarding skr mini e3 2.0 and skr tft50 screen combination, it is related to communication between the skr components and marlin firmware. It is really strange that g code command is send from screen to main board but just get ignored by firmware.
In my case filament change and filament detection are working in Marlin mode but not in touch screen mode,and if you connect filament run out sensor to screen connector it triggers on screen but no feedback from printer.

dmarnel, 2021-06-04:
Many of the issues with tft50 are also fixed by using the debug firmware.

radek8, 2021-06-04:
He has been working on repairs of the M600 command on the TFT recently. I think it should already be functional on TFT.
The condition is the current FW TFT and Marlin night issue.
it is also necessary to enable the following parameters in Marin:

Možnosti podpory dialogu s hostitelem:
EMERGENCY_PARSER (v Configuration_adv.h)
SERIAL_FLOAT_PRECISION 4 (v Configuration_adv.h)
HOST_ACTION_COMMANDS (v Configuration_adv.h)
HOST_PROMPT_SUPPORT (v Configuration_adv.h)

Možnosti podpory M600 s hostitelem a nabídka (Un) Načíst: (Možnosti podpory dialogu s hostitelem jako předpoklad)
NOZZLE_PARK_FEATURE (v Configuration.h)
ADVANCED_PAUSE_FEATURE (v Configuration_adv.h)
PARK_HEAD_ON_PAUSE (v Configuration_adv.h)
FILAMENT_LOAD_UNLOAD_GCODES (v Configuration_adv.h)

You can turn M600 simulation on or off in the config.ini file
try what suits you.

SamppaD78, 2021-06-04:
Thanks. I will check settings and see if i can get it to work.

SamppaD78, 2021-06-05:
Hi. Checked the settings SERIAL_FLOAT_PRECISION 4 (v Configuration_adv.h) and HOST_ACTION_COMMANDS (v Configuration_adv.h) was disabled. All other settings are correct and I dont have debugging enabled,as someone suggested in a email. Now M600 unload/load/resume works correctly when checked manually.
Thanks for help Radek8
Rest of the issues that needs some thinking about
My filament dummy run out sensor stopped responding in marlin mode?..I need to check this,as it was working with 2.07 marlin
Emergency shut down, when activated,printed is halted, it freezes temp readings so i  dont know if hotend and bed is turned off
After turning printer off/on it displays correct descending temps..it would be nice to visually see that temps are going down after shut down is activated