bigtreetech/BIGTREETECH-SKR-mini-E3 #526: TCM Connection Error Only powers on through USB xyz motors don't work, bed and extruder won't heat up.

PaulieWalnuts1111, 2021-03-13:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2021-03-13:
I don't understand your problem.
When powered only via USB motors and heating does not work.
But that won't be your problem.

Auke76, 2021-03-13:
I think I have the same issue - I didn't check the motors yet, but the display remains off - only when I turn on the Pi.4 with OctoPi etc. which is connected via the Ender 3's USB port, the display will turn on and give this "TMC CONNECTION ERROR" on the lowest line of the display.

radek8, 2021-03-14:
Do you have 12V or 24V connected?
Do you use an additional 5V power supply module?
In what position do you have the NeoPWR jumper?

rcxrbt, 2021-03-24:
That SKR board?
Try to isolate 5V pin on USB cable.

deckape964, 2021-03-24:
Your printer is not powered up. The mainboard is being powered by the usb connection. Disconnect the USB and start the printer and then connect it