bigtreetech/BIGTREETECH-SKR-mini-E3 #527: Press OK to continue, endless loop

dmarnel, 2021-03-14:
When in "Touch" screen mode, and using the (Un)Load feature from the Menu screen, Loading filament causes a popup window to appear, asking the user to tap OK to continue. Tapping OK closes the window briefly but then it reappears with the same message over and over again, with no way out but to power cycle the printer.

Johnc660x, 2021-03-19:
Yup i suffer the same issue, i think this is more of a TFT35 issue?

sarvenn, 2021-03-19:
Would you try the latest TFT firmware and prior to that please check the config.ini file and configure your marlin accordingly. First update your marlin with the request enables (mentioned in the config.ini file) then update your TFT firmware and try again.
That was a widely common issue and solved with the fixes. You may find numerous topics and pull request with a little research in the touchscreenfirmware repository.

dmarnel, 2021-03-20:
@sarvenn, I already had the latest TFT firmware installed (BIGTREE_TFT35_V3.0.26.x.bin), and Marlin 2.0.7.2, but I downloaded it anyway and rebuilt it after making sure all of the recommended enables from the TFT config.ini were set. 

The Un(load) option for loading filament still causes the endless loop popup asking the user to press Ok as stated in my original comment.

sarvenn, 2021-03-22:
Could you please update your Marlin to bugfix 2. There is a fix for this issue inside bugfix 2 firmware. It is related with the STM32 processor of skr mini board and issue is fixed with bugfix.

https://github.com/MarlinFirmware/Marlin/pull/21210 check this.

dmarnel, 2021-03-23:
@sarvenn, building from the bugfix branch worked. Thanks for the help!

closed 2021-03-23 by dmarnel