bigtreetech/BIGTREETECH-SKR-mini-E3 #531: [BUG] (Random Stepper settings driving the to hot)

john-clark, 2021-03-31:
### Description

Randomly the board will drive the steppers at the wrong settings.

### Steps to reproduce

1. Print
2. touch steppers
3. Burn fingers

**Expected behavior**
For the same settings to be used all the time

**Actual behavior**
Randomly the board will drive the steppers at the wrong amps and burn them up

### Additional Information
SKR 1.2 is terrible!

iZz0s, 2021-04-02:
Are your using mini 1.2, 2.0 or e3 turbo?

john-clark, 2021-04-02:
BIGTREETECH SKR Mini E3 V1.2 32bit Control Board + TFT35 E3 V3.0 Touch Screen Compatible TFT35 12864LCD Display WiFi 3D Printer Parts for Creality Ender 3 

using **https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V1.2/firmware-bltouch.bin**

Most of the time it works fine... one a month every 200 prints or so the steppers will run extremely hot. If I power it off for 5 minutes and let it cool down it will start working again. I have had to replace 1 stepper already.