bigtreetech/BIGTREETECH-SKR-mini-E3 #538: [BUG] Powering off and leaving the Raspberry Pi powered up, and USB cable connected kills Z Motor.

Anon0ne, 2021-04-20:
### Description

### Steps to reproduce

1. Power off the power supply
2. Leave the Raspberry Pi with Octoprint connected
3. There is some cause killing the Z Motor (I have burned up two in 2 months)

**Expected behavior**
The board does not have supply voltage, therefore the serial connection should not be able to control or do anything with motors. All that should be able to do is program the board.

**Actual behavior**
I was unable to measure current on the Z axis and voltage, but now since the motor is already dead again I will try.

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

bojanpotocnik, 2021-04-23:
Motors are powered (TMC2209 pins `VS`) from the `VBB` rail, which is main power connector 12/24V input via 20A fuse. USB power is `USB 5V` routed to `+5 V` via schottky diode. 

Therefore, motors itself are not powered via USB at all. But you said that changing the motor helps anyway, so TMC drivers are OK? Can you measure the resistance between all pins of "dead" and new motor? What is the difference?

Anon0ne, 2021-05-13:
> Motors are powered (TMC2209 pins `VS`) from the `VBB` rail, which is main power connector 12/24V input via 20A fuse. USB power is `USB 5V` routed to `+5 V` via schottky diode.
>
> Therefore, motors [...]

The meter I have in my office doesn't have the range i require for the entire motor pin i think. If i cut out one pin I am around 5.8 ohms on the old motor. If i turn the motor, it goes to 0.

I am awaiting new motor to test.

xHirscHx, 2021-05-31:
I'm having similar prob, when i power off but leave usb connected  to octoprint running  on pc when i turn power back  on and home my plate, the plate moves really slow then throws a code and shuts down the printer. new board Just installed yesterday

thebiggestgit, 2021-06-30:
Did you block the USB 5v line as detailed on Octoprint forum?   https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574

Also found this was the case with the stock creality board I had on my ender 5 pro with a USB cable and raspberry pi (4), it made a high pitched noise if the printer was turned off but the pi still was connected as well as slightly powering the original LCD panel.