bigtreetech/BIGTREETECH-SKR-mini-E3 #540: Unable to configure BLTouch 3.1 probe - Marlin Bugfix 2.0.7.2 - SKR Mini e3 V1.2

CraigVR85, 2021-04-28:
I'm at my wits end with this issue. I have recently installed the SKR Mini e3 V 1.2 on my Ender 5 - with a BLTouch.

If i use the precompiled firmware (from the SKR Mini e3 Github) the probe works perfectly without any issues. the problem is that I have reasons to want to alter the FW (linear advance and other features).

I have tried everything to configure the BLT in marlin 2.0 using the SKR mini e3 config files from the Marlin Configurations Github.

The probe fails to deploy when homing and i have to kill the power to prevent the nozzle from crashing into the bed. I've trawled the internet to find a solution, and while there are instances of similar failings none of these fixes seems to work for me.

Im convinced that it is a software issue (as i said the precompiled binaries (firmware-ender5-bltouch-for-z-homing-v1.2 works perfectly). 

I've enabled #define BLTOUCH, #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN, and Z_SAFE-HOMING - the problem seems to be the deployment, but i cant tell.

It is worth noting that with my marlin configs the probe does deploy and retract twice when the power is turned on. If i click the "deploy" and "stow" buttons repeatedly from the LCD the probe will occasionally deploy and stow - this is not repeatable.

Is there any way of getting the actual source files used to create the precompiled binaries? 

Any assistance is better than where i am now.

Regards

mikensan, 2021-05-26:
Are you using only the probe or probe + separate z switch? And if you're only using the probe version 3.1, the default wiring does not allow you to connect it to the z min pin... all 5 wires are in a single connector.

orrgii, 2021-06-09:
Had the same issue with the probe with a separate z-switch.  The crashing for me was due to the wires being reversed. Swapped and now I have to figure out how to stop it from printing 1/2 inch about the pad :-)

daniframinan, 2021-07-03:
Hi, I have the exact problem, but the precompiled firmware doesn't work for me, could you share the exact precompiled firmware you used?

CraigVR85, 2021-07-03:
I eventually used the ender 3 firmware and configured it for the 5. There’s a problem with that firmware config that I couldn’t find. Use the e3 and change the config.