bigtreetech/BIGTREETECH-SKR-mini-E3 #545: M25 causing crash

tacticalbacon89, 2021-05-20:
Using M25 during Cura Start Gcode crashes the printer 

the purpose i was using the M25 command was to first present the bed out and then pause the printing process right before the print actually begins so i can prep the bed with glue stick ext.. while its still hot and when im done i would just hit ok and it would resume that was my idea 

but..

 M25 activates, printer stops,  'message to ''press ok to continue'' prompts as it should  but when i hit the ok button the message keeps re prompting and print process does not move forwarded to next command

im using 
SKR Mini 2.0 E3
my firmware is the precompiled firmware from bigtreetech for the BL touch plugged into Z stop 
i also have the TFT35 touch screen
my firmware is on the SD card that came with SKR mainboard and it is left in the printer in the Main board SD slot
my Print File is being used on a Flash drive plugged into the TFT35 usb port

radek8, 2021-05-24:
Update FW TFT and FW Marlin 2.0.8.1to the latest.
In older versions there was a problem with the M25
If you set the required parameters Marlin as described in the instructions on the title page. It should work properly.
Instructions:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

a similar problem was solved here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1523

new FW TFT
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

tacticalbacon89, 2021-05-24:
> bigtreetech/BIGTREETECH-TouchScreenFirmware#1523

Do you have the same up to date version with BL touch pre compiled in