bigtreetech/BIGTREETECH-SKR-mini-E3 #546: tft35 and skr mini e3 v2.0 Simulation mode don't work

ChevalierLouis, 2021-05-24:
Hello

I would like to know which screen did I use on the marlin firmeware for the skr mini e3 v2.0 with the tft35 

I can use the screen in grafical mode but not with the simulation  

I have try with only the 5pin plug ==> only the grafical interface work 
I have try with only the 10pin plug ==> only the grafical interface work 
I have try with both plug ==> only the grafical interface work 
but never the simulation mode work 

have a nice day !!!

radek8, 2021-05-24:
What version of TFT 35 do you have?
How Are You Involved? Send a photo of the connection.
What have you configured in Marlin and TFT? Submit configuration files for review.

ChevalierLouis, 2021-05-25:
_What version of TFT 35 do you have?_
==>I don't know, but I bought it this month there is the new ui on it I will check it this evening

_How Are You Involved? Send a photo of the connection._

[image]
 
_What have you configured in Marlin and TFT? Submit configuration files for review._

==> I put //louis for each modification
[configuratio.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/6537062/configuratio.h.txt)

radek8, 2021-05-25:
You can at least take a picture of the display from the connector side

ChevalierLouis, 2021-05-25:
Of course 
[image]

ChevalierLouis, 2021-05-25:
Of course
[image]

radek8, 2021-05-25:
According to the photo you have the TFT35MZ
And the problem with not displaying Marlin mode is in a poorly connected 10Pin cable.

[image]

radek8, 2021-05-25:
[image]

ChevalierLouis, 2021-05-25:
Thanks a lot , I am so stupid;) It work

radek8, 2021-05-25:
Unfortunately, without an encoder only through the display, Marlin will be difficult to control. The right part of the display simulates an encoder, but it is not ideal.

radek8, 2021-06-12:
Can you close the ticket if your problem is solved?
Thank you

closed 2021-06-16 by ChevalierLouis