bigtreetech/BIGTREETECH-SKR-mini-E3 #547: [FR] (Marlin 2.0.8.2 update)

cookins, 2021-05-31:
hey
its new marlin out. when new updated version will be for e3 v2?
thanks

radek8, 2021-06-03:
As soon as you compile it :-)

fajnet, 2021-08-18:
Is it planned to support the new version of MarlinFW 2.0.9.1 for SKR mini E3 boards (v.1.0, v1.2 and v2.0)? When I download the new MarlinFW from MarlinFW.org, the platformio.ini file is incorrect and it lacks compilation methods for SKR boards.

radek8, 2021-08-18:
SKR mini E3 has no problem with compilation.
Just define the platform (chipset) in palatformio.ini and the compilation should be fine if you do not have a problem in your configuration.

radek8, 2021-08-18:
The SKR mini E3 uses the STM32F103CT6 processor

so it should be enough to overwrite the line
default_envs = mega2560
on
default_envs = STM32F103RC_btt_maple

cookins, 2021-08-18:
its already there. so lets wait for 2.0.9.1

closed 2021-08-18 by cookins

fajnet, 2021-08-18:
I see to Github SKR mini E3 repository, Firmware folder. Boards with version 1.0 and 1.2 have Marlin with version 2.0.7.x and board v2.0 have Marlin 2.0.8.x. Newest MarlinFW on Github MarlinFW is on version 2.0.9.1. On examples configurations this version is only Configuration.h and Configuration_adv.h files. On Marlin repository in platformio.ini file is not defined env for more boards (as SKR boards etc.). Is funkcionality platformio.ini files from this Repository SKR mini E3/firmware/v1.2/ to newest Marlin version 2.0.9.1 true? Thanks and Best regards

radek8, 2021-08-18:
I wouldn't rely too much on Bigtreetech precompiling Marlin.

It is always better to compile your own Marlin and be able to debug your own configuration.

radek8, 2021-08-18:
It's nothing complicated
Download Marlin 
https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip
Overwrite the above line in Platformio.ini
Transfer settings from the Configuration.h and Configuration_adv.h files
Adjust other settings according to your configuration (leveling probe, etc.)
Compile your own FW in VisualStudioCode

cookins, 2021-08-19:
> I wouldn't rely too much on Bigtreetech precompiling Marlin.
> 
> It is always better to compile your own Marlin and be able to debug your own configuration.

You should. They kinda always doing it...