bigtreetech/BIGTREETECH-SKR-mini-E3 #549: Marlin to 2.0.8.2.x compiled version breaks BLTouch

MRR0GERS, 2021-06-11:
BL Touch no longer functions properly after latest compiled firmware release.  On boot machine gives errors of "invalid mesh" and
"failed to enable Bed Leveling."

BL Touch tests fine from menu but machine will no longer auto home or even home Z.  Will also not allow to set ABL active.

Cheo7811, 2021-06-11:
Have you reset the EEPROM? Can you home Z alone? When you turn on the machine, does it test the BLTouch? it would help if you post you Configuration.h and Configuration.adv.h

MRR0GERS, 2021-06-11:
@Cheo7811 yes, I've reset the EEPROM; that was the first thing I did.  No, I can't even home Z by itself after the update.  The BLTouch does do the test on boot but it only does 2 instead of 3.  I didn't compile it myself, I used the precompiled binary, firmware-bltouch-for-z-homing.bin.

MRR0GERS, 2021-06-11:
I've reverted back to 2.0.7.2 for the meantime; that version is working fine.

Cheo7811, 2021-06-11:
What printer do you have? you just updated the firmware and nothing else? have you changed anything? cabling? what display do you have?

radek8, 2021-06-12:
If you use a precompiled bin file, it will be difficult to find the problem.
You have to compile your own FW. You can then tune and improve it.

bartchoo, 2021-06-13:
I have the same problem. Where can i find the older version for this? i'm pretty new at this all

chick911t, 2021-06-15:
Just like bigtreetech not to test same here bin file is not compiled correctly bltouch doesn't work I still have the old bin file which works fine

bartchoo, 2021-06-15:
> 
> 
> Just like bigtreetech not to test same here bin file is not compiled correctly bltouch doesn't work I still have the old bin file which works fine

Can i find the pre compiled firmware somewhere? i'm not so good in compiling it on my own.

chick911t, 2021-06-18:
> > Just like bigtreetech not to test same here bin file is not compiled correctly bltouch doesn't work I still have the old bin file which works fine
> 
> Can i find the pre compiled firmware somewhere? i'm not so good in compiling it on my own.

I have the older firmware-bltouch-for-z-homing.bin if you need it I am not sure if I am allowed to post link to Google drive but if you want to send a message I am on reddit user name is ender3po

AlexKlo71, 2021-06-19:
BL Touch no longer functions properly after latest compiled firmware release. On boot machine gives errors of "invalid mesh" and
"failed to enable Bed Leveling."

revjkramer, 2021-06-22:
@chick911t I could use a copy of the old firmware for z homing as well.

Ender3p0, 2021-06-26:
> @chick911t I could use a copy of the old firmware for z homing as well. It is kind of ridiculous they don't keep the old versions available for download. 
Old version of btts firmware 
https://drive.google.com/drive/folders/1FvddmlXX8Rq_gdPK0xO0Mjw-KXXxxJgE?usp=sharing

Ender3p0, 2021-06-26:
> @chick911t I could use a copy of the old firmware for z homing as well. It is kind of ridiculous they don't keep the old versions available for download. Here is the old pre compiled Btts firmware https://github.com/chick911t/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0

RaphaelHoefer, 2021-07-16:
@chick911t I would need a copy of version  2.0.7.2 too, if you could prodive :) Thanks!

MRR0GERS, 2021-07-16:
For 2.0.7.2 firmware files, you can go to https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/d5fc619c2a4aadc889c47680af351310f521e850/firmware/V2.0 and download them from there

Spensio, 2021-07-27:
> For 2.0.7.2 firmware files, you can go to https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/d5fc619c2a4aadc889c47680af351310f521e850/firmware/V2.0 and download them from there

At this firmware didn;t work well fan... must click menu to work. Maybe can some upload good file firmware.bin for Ender3 Pro Skr mini v2 for bl touch z home?

radek8, 2021-07-27:
Once you find out that the best compilation is your own, you will regret the lost time spent searching the web, looking for a bin file that would match your configuration. :-)

Take a look at 2-3 tutorials on youtube.
Download a sample configuration for your machine, customize it and compile.

No1sBoy, 2021-08-03:
BL Touch doesn't work well with mesh. Comment it out and choose bilinear. 

If you need working firmware files, Ivader has done a great job compiling binaries for all the BTT boards and loaded them all with all of the varying add ons in seperate files.
[](https://github.com/kad/Marlin/branches)
This will get you in the right direction. You can just download the source files and open the whole folder in VS Code and pick and choose which firmware you want to build...