bigtreetech/BIGTREETECH-SKR-mini-E3 #550: TFT35 V3.0 Error writing to EEPROM

TheGunPsy, 2021-06-12:
**Is your feature request related to a problem? Please describe.**
I am getting a message on the TFT35 screen. "Error Writing to EEPROM" when trying to save to EEPROM or typing M500 in the terminal. I am running the SKR E3 Turbo board. 

**Describe the solution you'd like**
I would like the LCD screen to write to the EEPROM.

Cheo7811, 2021-06-12:
I guess you are compiling Marlin, on Configuration.h enable the EEPROM,  #define EEPROM_SETTINGS and compile again.

TheGunPsy, 2021-06-12:
> I guess you are compiling Marlin, on Configuration.h enable the EEPROM, #define EEPROM_SETTINGS and compile again.

This is enabled already. Let me see if I can post my config&adv.h files

TheGunPsy, 2021-06-12:
Get hub is like learning a new language. I have posted my files, let me know if you can open them. 

Config.h: https://github.com/TheGunPsy/SKR-E3-Turbo/blob/main/Configurationh.txt
Config_adv.h: https://github.com/TheGunPsy/SKR-E3-Turbo/blob/main/Configuration_advh.txt

radek8, 2021-06-12:
What is the response to the command
M503
and
M504

TheGunPsy, 2021-06-12:
> What is the response to the command
> M503
> and
> M504

M503 returns a bunch of echos.... Looks successful. 

M504 says "EEPROM version mismatch(EEPROM=?MarlinV82)ok"

TheGunPsy, 2021-06-12:
Also here is the main issue in ponterface: 
SENDING:M501
echo:EEPROM version mismatch (EEPROM=? Marlin=V82)
echo:Hardcoded Default Settings Loaded

radek8, 2021-06-12:
EEPROM has a limited number of writes, after which it can fail and damage.
The EEPROM may be corrupted from frequent writes.
Maybe I would try to simulate the EEPROM on the SD card if the problem will recur.

radek8, 2021-06-12:
Can you read the description on the EEPROM chip?
If the correct size is defined.

radek8, 2021-06-12:
[image]

radek8, 2021-06-12:
[image]

TheGunPsy, 2021-06-12:
OK, I found the solution. 

under the following file: 
Marlin>SRC>HAL>LPC1768>HAL.h

comment out:
// i2c uses 8-bit shifted address
//#define I2C_ADDRESS(A) uint8_t((A) << 1)

Even thought the E3 Turbo board uses the LPC1769 chip use the HAL file under LPC1768. Once you recompile and flash, it will work in ponterface. 

After these changes M504 returns EEPROM ok and M501 will populate all machine parameters from EEPROM. The specific steps I took: 

1. Connect to ponterface
2. Send M502
3. Send M500
4. Begin work with any command. 

I will post the HAL.H file so you can inspect the code for those who are interested. 

/**
 * Marlin 3D Printer Firmware
 *
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 * Copyright (c) 2016 Bob Cousins bobcousins42@googlemail.com
 * Copyright (c) 2015-2016 Nico Tonnhofer wurstnase.reprap@gmail.com
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * HAL_LPC1768/HAL.h
 * Hardware Abstraction Layer for NXP LPC1768
 */

#define CPU_32_BIT

void HAL_init();

#include <stdint.h>
#include <stdarg.h>
#include <algorithm>

extern "C" volatile uint32_t _millis;

#include "../shared/Marduino.h"
#include "../shared/math_32bit.h"
#include "../shared/HAL_SPI.h"
#include "fastio.h"
#include "watchdog.h"
#include "MarlinSerial.h"

#include <adc.h>
#include <pinmapping.h>
#include <CDCSerial.h>

// i2c uses 8-bit shifted address
[... 133 lines cut ...]
 * set_pwm_frequency
 *  Set the frequency of the timer corresponding to the provided pin
 *  All Hardware PWM pins run at the same frequency and all
 *  Software PWM pins run at the same frequency
 */
void set_pwm_frequency(const pin_t pin, int f_desired);

/**
 * set_pwm_duty
 *  Set the PWM duty cycle of the provided pin to the provided value
 *  Optionally allows inverting the duty cycle [default = false]
 *  Optionally allows changing the maximum size of the provided value to enable finer PWM duty control [default = 255]
 */
void set_pwm_duty(const pin_t pin, const uint16_t v, const uint16_t v_size=255, const bool invert=false);

// Reset source
void HAL_clear_reset_source(void);
uint8_t HAL_get_reset_source(void);

inline void HAL_reboot() {}  // reboot the board or restart the bootloader

// Add strcmp_P if missing
#ifndef strcmp_P
  #define strcmp_P(a, b) strcmp((a), (b))
#endif

closed 2021-06-12 by TheGunPsy

radek8, 2021-06-12:
You're probably using old versions of Marlin.
These lines are no longer in the new version of the Marlin bugfix

TheGunPsy, 2021-06-13:
> EEPROM has a limited number of writes, after which it can fail and damage.
> The EEPROM may be corrupted from frequent writes.
> Maybe I would try to simulate the EEPROM on the SD card if the problem will recur.

@Radek8, I found the solution. It is listed above your response I am replying to. I cant thank you and everyone else enough for your responses and dedication to problem resolution! Thank you very much!

Cheo7811, 2021-06-13:
Sorry, I don't get it, what was the solution?

radek8, 2021-06-14:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/issues/30