bigtreetech/BIGTREETECH-SKR-mini-E3 #553: Z axis will not home - SKR mini E3 v2 w/bltouch

CamWynd, 2021-06-25:
### Description

Z axis homing with bltouch sensor is not working properly.

Replaced a dead SKR minie3  V1.3 board with a new V.2 board. the BLtouch does it's little dance test, and I can trigger testing through the touch panel. I can 'drive' all axis through the touch panel properly. When I tell the printer to home, the X and Y axis home properly, but the Z axis moves to the center of the bed, goes up, deploys the bltouch needle and then probes in mid air for the bed (still cm below the probe). 

### Steps to reproduce

1. wire in new board
2. compile and install new firmware.bin 
3. request all 3 axis to home 
4. X and Y home, Z does not

**Expected behavior**
I expect the Z axis to home to the print bed and have the BLtouch sensor stop the process as it encounters the print bed (as it did under the V1.3 board)

**Actual behavior**
X and Y move to the center of the print bed, and the Z axis rises up before deploying/stowing the BLtouch, then reports an error from the BLtouch and stops.

[configs.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/6712896/configs.zip)

CamWynd, 2021-06-25:
Read that this could be a cabling issue, so I replaced the cable. Got the same problem.

radek8, 2021-06-25:
send photos of BL-Touch connection

radek8, 2021-06-25:
Try these changes.
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/6719047/Configuration.zip)
[Configuration_adv.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/6719105/Configuration_adv.zip)

CamWynd, 2021-06-26:
Thanks, but those configs give the same result. Looks like the printer thinks it's z endstop is waaay above the print bed, so my next thing is researching how to change it's opinion.

Ender3p0, 2021-06-26:
Try defining //#define BLTOUCH_FORCE_SW_MODE in configuration adv

CamWynd, 2021-06-26:
Tried the #define BLTOUCH_FORCE_SW_MODE and got the same result.

Today I disconnected the BLtouch and reconnected the stock Z stop sensor and everything worked correctly. Next up I replaced the BLtouch cable and returned the sensor to the printer. Problem is back. So . . . the BLtouch sensor is baffed? Except it worked fine under the old SKR-mini-E3 board (no longer alive)? I've seen so many posting about the BLtouch/V2.0 board I'm starting to wonder if there's something wrong deeper in the code . . .

CamWynd, 2021-06-27:
Got it working. 
Seems the Z-probe interface isn't ready for prime time. I moved the 2 pin B&W cable over to the Z-stop interface and recompiled to use for the BLtouch. Works like a charm.

closed 2021-06-27 by CamWynd