bigtreetech/BIGTREETECH-SKR-mini-E3 #554: Bltouch installed correctly and screen is black.

gtdaniel101, 2021-06-25:
I have installed a bltouch to my ender 3 pro that has a skr mini 1.2 motherboard. I flashed the firmware and plugged in the bltouch with the orders being blue/brown, red, yellow. Then black, white. Whenever I turn the power on, the fans will turn on but the screen wont. If I unplug either the blue or red wire the screen will turn on but I lose the ability to control the bltouch at all.

radek8, 2021-06-25:
Looks like you have the BL-touch wires connected in the wrong order.

radek8, 2021-06-25:
send photo of engagement

gtdaniel101, 2021-06-25:
[image]

gtdaniel101, 2021-06-25:
then the black and white ones are plugged into the ztop

CamWynd, 2021-06-25:
I had this problem too. I had two wires backwards in the plastic connector; what colour wires does your unit have?

gtdaniel101, 2021-06-25:
blue red yellow black and white

radek8, 2021-06-25:
bad photo
send new photos to see where all the BL-Touch wires are connected
Also take a picture of the BL Touch to issue a color sequence.

I don't see a connected display anywhere. it cannot work if it is not connected.

radek8, 2021-06-25:
To which connector is black and white connected?
PROBE or Z-STP?

radek8, 2021-06-25:
Submit configuration files

gtdaniel101, 2021-06-25:
[image]
I unplugged the display just so I could take the picture, the wires were covering it up. The black and white are connected to the Z-STOP

gtdaniel101, 2021-06-25:
[image]
Hopefully, this is a better picture, I don't have a 3pin plastic piece so I just used a two-pin and put the third in, no wires touch so it should be fine.

radek8, 2021-06-25:
Then you have it completely badly connected

gtdaniel101, 2021-06-25:
Sure it isn't pretty but electrically everything works, nothing is touching, everything is separated by plastic and each pin has a solid connection to the board.

radek8, 2021-06-25:
The colors of the cables on the BL -touch are folded back

gtdaniel101, 2021-06-25:
What do you mean by that? I don't understand

radek8, 2021-06-25:
[image]

radek8, 2021-06-25:
[image]

gtdaniel101, 2021-06-25:
[image]
Thank you so much, I changed them around and it is working now. I was going off of this picture and multiple others that showed the same thing for what the wires were

closed 2021-06-25 by gtdaniel101

radek8, 2021-06-25:
You came up with the right picture, but your connector has poorly folded colors (in the opposite direction)