bigtreetech/BIGTREETECH-SKR-mini-E3 #555: [BUG] new bin file BLTouch-zhoming incorrect

Ender3p0, 2021-06-26:
After flashing bigtreetech’s pre compiled firmware firmware-bltouch-for-z-homing.bin BLTouch no longer works correctly when you  turn on machines BLTouch dose it’s test sequence and errors, all wires are correct as older btt pre compiled firmware works correctly

CamWynd, 2021-06-26:
Are you using the V.2 board as well? I reconnected the Z stop switch on mine and everything works fine. Reconnected the bltouch and Z homing stopped working.

Ender3p0, 2021-06-26:
> Are you using the V.2 board as well? I reconnected the Z stop switch on mine and everything works fine. Reconnected the bltouch and Z homing stopped working.

Yes the the v2 didn’t reconnect the z stop switch as I was using the BLTouch for homing. Now you have got me thinking the may be mixed up on GitHub when I get a chance I will have to flash the bltouch bin file and see what that dose

CamWynd, 2021-06-27:
Got mine working. Seems the Z-probe interface isn't ready for prime time. I moved the 2 pin B&W cable over to the Z-stop interface and recompiled to use for the BLtouch. Works like a charm.