bigtreetech/BIGTREETECH-SKR-mini-E3 #556: cannot compile, error The terminal process "...\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1"

Pixelluca, 2021-06-29:
### Description

I downloaded the latest master code, and wanted to enable bltouch, z-homing, and change the offset. However, whenever I try to compile the firmware, i get this error: `The terminal process "...\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1"`. I've noticed that there are multiple errors in the code, all of which say that `function call is not permitted in a constant expression`. I have noticed that all of these errors are on the word `ENABLED`, and under an `#DEFINE` that has been commented out. I tried to comment out the faulty lines, as they appeared to not be necessary to me, but the errors did not go away. I do not have a lot of knowledge of programming, so I don't know any other solutions.

### Steps to reproduce

1. Compile code.

**Expected behavior**
The firmware should compile.

**Actual behavior**
The firmware doesn't compile and shows error `The terminal process "...\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1"`.

### Additional Information
* I have included a .zip of my configuration.h
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/6731991/Configuration.zip)

RetArsonFD, 2021-07-03:
I've been getting very similar errors also when trying to compile the firmware.  I have no clue what to do.

travisclare, 2021-07-29:
Having same problem

Santravis, 2022-01-25:
Having same problem

RetArsonFD, 2022-01-26:
> Having same problem

FWIW.....I have learned that ANY little difference in the "name" of the file will throw an error.  The 2 big (and simplest to fix) are 1) any spaces in the line  (ex.  happy portfolio.  Has to be happyportfolio), and 2) to long of a file name.  I'm not sure what the limit is, but I know they can be too long.