bigtreetech/BIGTREETECH-SKR-mini-E3 #559: [BUG] Crashes when executing G28(with BLTouch)

TheRealRazr, 2021-07-07:
I'm not sure which version of BLTouch I'm using but I've got the BTT SKR mini E3 v1.2.  When I execute G28 or just go through Marlin and start Auto home it homes the X and Y axis but when it starts homing the Z axis my BLTouch deploys and retracts and I get message saying there was a BLTouch error in Octoprint, In Marlin it says the printer has stopped. I have USE_PROBE_FOR_Z_HOMING enabled.

If anyone has any clues to what's causing it and how to fix the issue I would greatly appreciate it as I'm eager to start using my printer :\

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**
Too home the z axis with Probe

**Actual behavior**
Crashes.

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
[configuration_h.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/6780550/configuration_h.zip)
* Provide pictures or links to videos that clearly demonstrate the issue.
[image]

TheRealRazr, 2021-07-08:
Ok so now I've tried flashing the pre-configured firmware.bin file for bltouch and I am still having the same issue, it will home the X and Y axis but crashes at the start of Z homing.

Is this a hardware problem? To clarify, my bltouch was working perfectly before, it had no problems homing yesterday until the end of the day when I was adjusting the Probe Z offset, after that it took a dump and wouldn't home the Z axis.

TheRealRazr, 2021-07-08:
I'm really convinced now that this is a hardware problem. The cheap Chinese made pin connectors on the bltouch are having a hard time staying connected to the motherboard I think. I'm gonna try using hot glue or something for it to keep a good connection.

chick911t, 2021-07-08:
Hi the new bin files with bltouch don't work, sorry bltouch doesn't work be for you blame the bltouch here is the old btt bin files for the ender https://github.com/chick911t/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0

TheRealRazr, 2021-07-08:
> Hi the new bin files with bltouch don't work, sorry bltouch doesn't work be for you blame the bltouch here is the old btt bin files for the ender https://github.com/chick911t/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0

Thank you for the help, I'll be sure to have a look at that. though after playing around with my printer last night I am fairly convinced now that this is a hardware problem I'm having, Possible with the pin connecters on the Bltouch for the z-stop pins. It seems that when it stops working and it can't home the z axis I just have to wiggle the z-stop wire a little to make sure it's making a connection, and then after restarting my printer and homing it starts working.

closed 2021-07-09 by TheRealRazr