bigtreetech/BIGTREETECH-SKR-mini-E3 #563: [BUG] 3.1 bltouch (z stop configuration) wont home Z error M999

beepbeepMF, 2021-07-16:
I have installed the 3.1 Bltouch onto an skr mini e3 v2/TFT ender 3 pro. I flashed the correct firmware from https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0 according to the z stop wiring configuration.  The printer will fail ABL and state "home z axis"
When z axis homing is attempting it moves the the middle of the bed, the bl touch moves up and down (z does not move) and the error shows restart with M999
I thought at first it was the BL touch or wiring so I have changed out both and have encountered the same error.

Marvin41, 2021-07-16:
The same problem is also with me BTT SKR mini E3 v2.0 and Creality Ender 3 pro

beepbeepMF, 2021-07-17:
https://www.danbp.org/p/en/node/149

I used this firmware today and it has been working well so far. Be sure to read the info to make sure you are ok with some of his settings.

chick911t, 2021-07-20:
Yes the bin files seem to be incorrect here is the older version of btt pre compiled firmware it's marlin 2.0.7.2 but it works so you will be back up and running
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/d5fc619c2a4aadc889c47680af351310f521e850/firmware/V2.0

Spensio, 2021-07-27:
> https://www.danbp.org/p/en/node/149
> 
> I used this firmware today and it has been working well so far. Be sure to read the info to make sure you are ok with some of his settings.

After upload this firmware i can't connect printer with USB repeiter host or any other programm like cura etc. You too have same problem ? how to solve it. At main file bightreetech i didn;t have this problem.

anh-duc-le, 2021-08-05:
> Yes the bin files seem to be incorrect here is the older version of btt pre compiled firmware it's marlin 2.0.7.2 but it works so you will be back up and running
> https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/d5fc619c2a4aadc889c47680af351310f521e850/firmware/V2.0

Thank you so much, I have spent over 3 hours of checking the hardware and googling for answers. And the simplest answer was just to use an older version :)

chick911t, 2021-08-05:
> Thank you so much, I have spent over 3 hours of checking the hardware and googling for answers. And the simplest answer was just to use an older version :)

Lol sorry you don't find it sooner hopefully they get it right on there next update, I try and keep older working copies a on my computer just in case

Tonychazz321, 2023-06-07:
idk if you fixed this problem yet but i had the same m999 problem when installing my bl touch on a skr mini e3 v3. the black and white wire on the bl touch harness was my problem. had them plugged into the last two pins on the bl touch socket (where you'd think it would go) nope. needs to be wire to the z stop switch port if you're using the bl touch for z homing firmware.