bigtreetech/BIGTREETECH-SKR-mini-E3 #564: [FR] bigtreetech PLEASE use Tags for releases

MRR0GERS, 2021-07-16:
**Is your feature request related to a problem? Please describe.**
It is a bit time consuming to locate previous file/firmware versions. 

**Describe the solution you'd like**
Add a tag with each release to help as a pointer to specific versions.  This is really useful in case there is an issue with newer versions to be able to easily find, download and revert to another firmware.

**Describe alternatives you've considered**

**Additional context**

beepbeepMF, 2021-07-16:
Did you find 2.0.7? if so do you have a link to the firmware?

MRR0GERS, 2021-07-16:
> Did you find 2.0.7? if so do you have a link to the firmware?

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/d5fc619c2a4aadc889c47680af351310f521e850/firmware/V2.0