bigtreetech/BIGTREETECH-SKR-mini-E3 #566: [FR] (feature request title)

Medjeberabdou, 2021-07-30:
hi
can i used servomotor panasonic with bigtreetech skr e3 v2.0 !? and how !!

radek8, 2021-07-31:
If it's a standard servo, and you're not using touch, use the z-probe connector

Medjeberabdou, 2021-07-31:
my issue is want to know if this card bigtreetech skr e3 mini v2.0, can order with succe my own 3d printer, who i want made it with 3 axes used panasonic servo motor.
Cordially.

radek8, 2021-07-31:
What will the servo motor switch? 
Step motors are used for movement

radek8, 2021-07-31:
I'd rather pick SKR V1. 4 Turbo

Medjeberabdou, 2021-07-31:
This what is mine Panasonic servomotor with own driver (closed loop) 
https://www.panasonic-electric-works.com/ch/fr/servocontroleur-minas-a5.htm
So for 3 axes XYZ i will used this servomotor with own driver,   what do tout think for SKR v1.4 turbo Can give me step/dir in output.

radek8, 2021-07-31:
Be better next time.
By servo, I mean a servo, not a closed-loop stepper motor.
If you wanted to use the SKR mini E3 V2 (again, your comment is unclear), then it is completely unsuitable for your purpose, because it is already firmly equipped with the TMC2209 driver.
SKR V1,4 or SKRV1,4 Turbo are applicable for closed loop engine.

radek8, 2021-07-31:
[image]