bigtreetech/BIGTREETECH-SKR-mini-E3 #567: ender 5 skr eV2+TTF3.5 display + Bltouch

3dprinte, 2021-08-31:
does anybody have a pre compiled firmware for this particular set up im not to familiar w vs code this is the worse 
ender 5 +ttf3.5+bltouch 
tia>>> willing to pay someone to compile for me if not available just need help with this ={

Runk7381, 2023-01-17:
Having the same problems.... New board and display... 5 pin Bl touch installed...it powers up, but does display auto level error..Ender 5

rhweese, 2023-02-18:
Having the same problem. I've looked all over github for firmware for the Ender 5 firmware for the Btt skr MINI E3 v2.0+ TTF3.5 display + BLtouch and have not been able to find firmware anywhere. Bigtree Teck's skr MINI E3 v2.0+ TTF3.5 display + BLtouch and the skr eV2+TTF3.5 display + Bltouch  I here are great upgrades, but that being said, it is not if you con not find the correct firmware to use. The product at this point becomes nothing more than useless garbage. Big Tree Teck could help it's customers out with better customer service  if they would at the very least give a link for all printers available for this configuration. Ender5 and Ender 5 Pro  firmware configurations for this setup I have not been able to find anywhere. The Ender 5 Pro build (220,220,300) is the same as the Ender 5"S so I you can use this firmware as almost plug and play, any changes you want to make are very easy even for the beginner just getting their feet wet in Marlin. If I can find the firmware for the Ender 5 or the Ender 5 pro for the skr eV2+TTF3.5 display + Bltouch or the  skr MINI E3 v2.0+ TTF3.5 display + BLtouch i will add a link here to direct you to where it is located. If you find it please let me know in return. Happy hunting!!!

rhweese, 2023-02-19:
FOLLOWUP! See if this  link is any help for Ender 5 [user's.]  (https://github.com/alepee/Marlin-Ender-5). Read down the page and check the info for your platform to see if it is included. I hope this is of some help. It worked for me and my 2019 Ender 5.