bigtreetech/BIGTREETECH-SKR-mini-E3 #574: [FR] Marlin 2.0.9.2 official BTT firmware release

cookins, 2021-10-15:
Hey!
2.0.9.2 is out! I hope BTT will found time for new firmware :)
thanks!

jmsferGH, 2021-10-15:
Subscribe to that !!

laverdone, 2021-11-18:
Yes i'm too

cookins, 2021-11-18:
lets hope they not abandon 2 version with 3rd version out...

Eclipsegsxr, 2022-01-07:
I have compiled my own using the new Marlin and have the skr mini 3e v2 and tft touch screen and Official BLTouch 3.1. All working as expected thus far on my 3nder 3 pro. I have my BLTouch pulled into the board and all endstops are still in place. It probes a 25 point grid for ABL. probe is used to home z axis in center of the bed.

cookins, 2022-01-24:
HEY, BTT, please - its time to update your firmware for 2.0 E3! its 2.0.9.3 out! we still got 2.0.8.2...