bigtreetech/BIGTREETECH-SKR-mini-E3 #585: [BUG] SKR MINI E3 V3 - FANS

pacocatech, 2021-12-01:
Folks,
I bought a SKR MINI E3 V3.0 and I'm having problems with the FANS ....
What is the correct connection to make on this card?
I currently used the following connection without success:

FAN0: FAN that is on the board
FAN1: FAN that is in the hotend
FAN2: FAN that cools parts

When I start printing, the fan that cools the parts doesn't turn on and the fan that should stay on all the time to cool the plate is never turned on.

Can you help me?

radek8, 2021-12-01:
On what date do you have Marlin compiled?
Recently there was a problem with soft.PWM and fan control on STM32 platforms
https://github.com/MarlinFirmware/Marlin/issues/20638
https://github.com/MarlinFirmware/Marlin/issues/22972

fixed:
https://github.com/MarlinFirmware/Marlin/pull/23125

pacocatech, 2021-12-01:
I'm using marlin source shared by BTT:

https://github.com/bigtreetech/Marlin/tree/SKR-mini-E3-V3.0-G0B1

radek8, 2021-12-01:
That will be the problem.
The BTT version is 19 days old.
The fix was uploaded 8 days ago.
Compile your own FW.

pacocatech, 2021-12-02:
But the official version of marlin still doesn't support this new board.

radek8, 2021-12-02:
So that's bad luck

radek8, 2021-12-02:
https://github.com/MarlinFirmware/Marlin/issues/23189

pacocatech, 2021-12-02:
Thanks a lot man!
I'll wait for merge request have been approved.

Sensei1978, 2021-12-06:
Hello. I have the SKR Mini E3 V3 on an Ender 5 Pro. But the part cooling fan wan't to turn on.
Can anyone help me?

zuniguns, 2021-12-06:
> Hello. I have the SKR Mini E3 V3 on an Ender 5 Pro. But the part cooling fan wan't to turn on. Can anyone help me?

I have the same problem here. the board works well, but dont start any fan (doesn't even start the front fan that cools the heatblock) I checked with multimeter all the fan ports, and dont send any signal at all. Anyone help us :(

radek8, 2021-12-06:
Try the new versions. Is 4 days old and may have a repair ...
https://github.com/bigtreetech/Marlin/tree/SKR-mini-E3-V3.0-G0B1/Marlin

zuniguns, 2021-12-06:
Yeah! they lauched new firmwares 13h ago! Works for me, but now i have BL Touch Problems with this one.

H3R0Semmie, 2021-12-06:
> Yeah! they lauched new firmwares 13h ago! Works for me, but now i have BL Touch Problems with this one.

Where did you find the new firmware?

zuniguns, 2021-12-07:
> > Yeah! they lauched new firmwares 13h ago! Works for me, but now i have BL Touch Problems with this one.
> 
> Where did you find the new firmware?
> 
> Edit: Nvm found it, https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin

I was looking to send it to you :) Are you going to use the version with BL TOUCH Z HOMING? if so, let me know if it worked for you, because everything works for me, except the fans in the first firmware version, and in this second everything works (including the fans) but the bl touch doesn't work anymore.

emersonsc, 2021-12-13:
Has the fan and BLTouch issues been resolved?

BradisBowser, 2021-12-15:
Any luck on this one? BLtouch for Z homing isn't working for me at all. Hoping to stumble across a fix so I can get the new board up and running.

zuniguns, 2021-12-15:
> Any luck on this one? BLtouch for Z homing isn't working for me at all. Hoping to stumble across a fix so I can get the new board up and running.

the only way it worked for me was using this latest firmware version, but i had to change the black and white connectors from BL touch to the Z stop pin. through the BL touch port, no way!

Yathani, 2021-12-27:
I have the same issue where board fan always on.

Everything starts silent but once i move any of the axes, the board cooling kicks in and never turned off. I can turn it off manually inside the configuration but it won't do it by it self even if i set it the time to 10. Hotend fan turn on correctly above 50C as well as parts cooling from slicer setting or manually.

I plugged the board fan into fan 2 with pin# PB15. I have also compiled my own firmware basis Marlin 2.0.9.3.

The controller fan does turn off automatically if I use BTT latest firmware but not my own.

Not sure what i'm missing

Yathani, 2021-12-27:
> Folks, I bought a SKR MINI E3 V3.0 and I'm having problems with the FANS .... What is the correct connection to make on this card? I currently used the following connection without success:
>
> FAN0: [...]

The correct fans wiring are 
FAN0: Prints part cooling
FAN1: Hotend fan
FAN2: board cooling fan

Yathani, 2021-12-27:
I have managed to fix the issue on my Ender 5 and the fan now turns on and off accordingly using Marlin 2.0.9.3

The board cooling suppose to turn off after x seconds after the stepper motor is disable as per 
"#define DEFAULT_STEPPER_DEACTIVE_TIME" which is 60 second in my case but fan didn't turn off.

The issue was that "#define DISABLE_INACTIVE_Z" was set to false where all other are true causing the Z motor to be active after homing and won't be disabled. Simply set the inactive_z to true as well.

vesami, 2022-01-03:
> 
> The correct fans wiring are FAN0: Prints part cooling FAN1: Hotend fan FAN2: board cooling fan

Thanks, I used intructions from makenprint.uk* and they had fans wired incorrectly.

*https://www.makenprint.uk/3d-printing/3d-printing-guides/3d-printer-mainboard-installation-guides/btt-skr-mini-e3-v3-guides/btt-skr-mini-e3-v3-setup-guide/#fans

Caleb79, 2022-01-05:
I checked the wiring accordingly but even with the latest Marlin version the hotend fan and the board fan are still not working. 
What can I do?

rohitz123, 2022-01-20:
Guys i am facing the same issue here and the fan is not working. 
can someone guide me how to fix it...

Monster-Yeti, 2022-01-22:
Im still facing the same problems even with the latest Marlin configured with Soft PWM on too,

Any one have any luck and fancy sharing a build for ender 3 pro with bltouch for z homing

emersonsc, 2022-01-24:
> Im still facing the same problems even with the latest Marlin configured with Soft PWM on too,
> 
> Any one have any luck and fancy sharing a build for ender 3 pro with bltouch for z homing

https://github.com/emersonsc/Marlin
Here are my config files. You'll need to adjust the properties for your thermistors, and advanced park to match your extruder.

Monster-Yeti, 2022-01-24:
> https://github.com/emersonsc/Marlin Here are my config files. You'll need to adjust the properties for your thermistors, and advanced park to match your extruder.

Much appreciated, really want to get back to printing 24/7 but not without the parts fan! really thought big tree would be on this.

srettew, 2022-06-21:
I’m joining this thread because I just installed an E3 V3.0 on my Ender 3 Pro.  This is regarding the 3 fan assignments and getting them to work correctly. I confirmed the following by checking function:

FAN0 (PC6) = Part cooling fan
FAN1 (PC7) = Hotend cooling fan
FAN2 (PB15) = Control board cooling fan

FAN0 turns on correctly by using fan speed control on TFT35
FAN1 turns on correctly when hotend temp is goes above 50 degrees and turns off when lower than 50 degrees
FAN2 is the problem. This is supposed to turn on when a driver is activated.  I get nothing when I manually move any of the axis using the movement control on TFT35. 

Where in the config_h file can I find PB15 and what is the correct code fix to make it work?

zuniguns, 2022-06-21:
> Folks,
> I bought a SKR MINI E3 V3.0 and I'm having problems with the FANS ....
> What is the correct connection to make on this card?
> I currently used the following connection without success:
>
> FAN0: [...]

Do you have BL Touch installed? I used the second FW provided by BTT and all works fine. I dont upgraded this last FW, so idk how its working. If you want i share with you the FW of mine. (But its for BL Touch for Z homing)

Nbrooks639, 2022-06-22:
i am also facing these same problems on my ender 6 i can get board fan to turn on and work correctly. but the hot end and parts cooling fan will turn on together

StrawberryD24, 2022-07-31:
I have the V3 installed on my ender 3 and the part cooling does not work. Any fixes would be really appreciated

srettew, 2022-07-31:
You have to assign the fan pins in Marlin. I had same problem until I updated firmware and assigned them in VS Code.

StrawberryD24, 2022-08-02:
How do you go about that? 
I've tried editing these lines in config_adv.h both generate errors the first being that a board with support for
multiple fans is required which is the case with the V3???
The second line, I alreafy had uncommented which I believe defines the pin number for the controller board fan. This fan works as intended. 

#define REDUNDANT_PART_COOLING_FAN 2  // Index of the fan to sync with FAN 0.
#define CONTROLLER_FAN2_PIN FAN2_PIN  
Attempting to switch The controller fan pin to the redundant fan pin, which i'm assuming is FAN_PIN generates an error, that is not allowed lol.

"""" 2187 |     #error "You cannot set CONTROLLER_FAN_PIN equal to FAN_PIN.""""""

I got the pin numbers from the pins folder 

#define HEATER_0_PIN                        PC8   // "HE"
#define HEATER_BED_PIN                      PC9   // "HB"
#define FAN_PIN                             PC6   // "FAN0"
#define FAN1_PIN                            PC7   // "FAN1"
#define FAN2_PIN                            PB15  // "FAN2"

I am currently compiling the bugfix2.1.x version of marlin 
That is after giving up trying with the current stable release.

Just to reiterate, the controller fan and hot end fans work as intended. 
But the part cooling fan........ Oh Boii

404brainz, 2022-08-12:
Got my new board and 3d touch, hooked up with Zhoming, and I have the same fan issues as the rest.  My part cooler fan will not spin at all.  Hope Bigtreetech addresses soon, seems like an ongoing issue.

srettew, 2022-08-12:
You have to assign the pins for fans in firmware. There are no assignments in the version you download for V3.

StrawberryD24, 2022-08-14:
@srettew How do you assign these pins in marlin?

StrawberryD24, 2022-08-14:
I'm still on the same issue

StrawberryD24, 2022-08-14:
@srettew Can you share your modified config files or pin files ?

srettew, 2022-08-14:
To edit Marlin you need https://code.visualstudio.com/

StrawberryD24, 2022-08-14:
@srettew I have visual studio with platform io, marlin builder, c++, python and all the extra extensions. I'm able to successfully edit and compile marlin firmware. I just need to know which files to edit for the pin definitions. I have tried a few so far I've. Config.adv and stmg0 pins file. With no success.
So which lines do I edit from which file.
If you could share yours, I'm more than happy to compile myself

DevlshOne, 2022-08-25:
I have the same issue(s).

Only 1 FAN appears in Marlin Mode on the controller.
No PARTS COOLING FANS working - via M106 or from Marlin Mode.
Hotend fan does not come on with nozzle.
I do compile my own firmware.

Is REDUNDANT_PART_COOLING_FAN verified to be operational?  Do we know the CORRECT pins / headers to use for each fan?

StrawberryD24, 2022-08-26:
@DevlshOne I'm currently using bugfix 2.1 with both the controller fan and hot end fans working correctly that's Using the pins defined in the pins/stm32g0 file. Redundant part cooling fan does not let marlin compile when uncommented. Thus I've had a lot of trouble getting the part cooling fan to work on my ender 3. Do note that the pin numbers seem to be a bit inconsistent.

DevlshOne, 2022-08-26:
> with both the controller fan and hot end fans working correctly

I am now able to get those working, as well. Thanks. Still no parts cooling fans, though. And an error 'Board does not support enough PWM fans' when trying to use `REDUNDANT_PARTS_COOLING_FAN`.

I've had issue after issue with everything I've bought from @bigtreetech . How do they stay in business?

jkcab95, 2022-08-26:
Ok I had a v2 board that the mosfet burned up. I purchased a v3 board and installed it. I downloaded Bigtree firmware and it did not work with the TFT35 so I tried unsuccessfully to update the screen firmware because it required Marlin 2.1 or later. I then found that Bigtree does not have firmware for marlin 2.1. So I went to Marlin and downloaded the latest with configuration examples and configured my own from The version 3 example for the ender 3 pro. The controller fan comes on when the hot end reaches 50c? The hot end fan never comes on and the part cooling fan works fine. I went back to the configuration H adv and put in the appropriate pin numbers for the fans....FAN_0 PC6 Fan_1 PC7 and FAN_2 PB15. I recompiled and uploaded and the operation is the same?????? I checked and all fans work when powered but there is never any power from the board for the hot end fan and the controller fan works as if it were the hot end fan? I can not even test to see if the printer stops in the middle of the print from the old screen firmware as I can't run the printer without a hot end fan. Also as a side note you can not use a high heat thermister with this board and the latest marlin firmware (undetermined conflict). I had to change it to a standard thermister so it would register anything. I just want to be able to print something. The version 2 board had hardly any hours on it when the part cooling fan failed. My biggest complaint is I have a Bigtree H2 hot end a TFT35 screen and their board and they have no compatible firmware for their own products!

jules-keg, 2022-09-01:
> I am now able to get those working, as well. Thanks. Still no parts cooling fans, though. And an error 'Board does not support enough PWM fans' when trying to use `REDUNDANT_PARTS_COOLING_FAN`. [...]

I have the same issue, both fans work correctly after modifying Marlin Firmware. Cannot get the parts fan to turn on. Also cant find it anywhere in the Configuration.h or Configuration.adv.h. Anyone that has been able to solve it? I can guide anyone else to make the other fans work as they should!

srettew, 2022-09-01:
The pins file containing the references for this board can be found in the Marlin firmware, at
the following path: Marlin/src/pins/stm32g0/pins BTT SKR_MINI_E3 V3 0.h