bigtreetech/BIGTREETECH-SKR-mini-E3 #587: Skr mini e3 update firmware issue

KAMX89, 2021-12-04:
Hello i recently bought skr mini
I tryied to upload firmware via SDcard 
After restart with card inside second Led blinks, and then stop and then both glow. 
Also it stopped being detected in Windows on connected USB cable.
Does it bricked or what?
When i try to upload other firmware it ends same. Longer or slower depends on filrsize. 
What i can do?

radek8, 2021-12-05:
Will the FW file be renamed? 
From bin to cur?

KAMX89, 2021-12-05:
Yes it is renamed. But still it doesnt detected in Windows.

radek8, 2021-12-06:
Do you have a USB port enabled in Marlin?

radek8, 2021-12-06:
If the FW file is renamed, it was encountered successfully. So your problem is only in the configuration.

KAMX89, 2021-12-06:
I tryied different lcd different firmware. Nothing work..

radek8, 2021-12-07:
Build your own FW