bigtreetech/BIGTREETECH-SKR-mini-E3 #589: SKR mini E3 V2.0 problem with BLTouch for z homing

IWantAnAccount, 2021-12-05:
There is a problem with auto bed leveling when using the BLTouch.
The problem is that the print head goes into position, but the BLTouch repeatedly just pulls the pin in and out. On the Z-axis the position does not change, which is why this all happens in the air and the attempt only ends with an abort.
 
In addition, the measuring points are too far out on the print bed for the standard BLTouch extension for the Ender 3, so the points on the left of the print bed cannot be reached properly by the Z probe, if at all.

Board: SKR mini E3 V2.0
Printer: Ender 3
Z probe: BLTouch
firmware: firmware/V2.0/firmware-bltouch-for-z-homing.bin

With older versions I only had the 2nd problem of these two

zuniguns, 2021-12-08:
When you send the comand for abl or home in all axis the printer send home in x and y axis, but when the printer head reach the center to measure the z axis, the BL touch just send deploy and stow repeatedly and not Go down the z axis?

IWantAnAccount, 2021-12-15:
> When you send the comand for abl or home in all axis the printer send home in x and y axis, but when the printer head reach the center to measure the z axis, the BL touch just send deploy and stow repeatedly and not Go down the z axis?

exactly that

marc-gist, 2021-12-17:
> > When you send the comand for abl or home in all axis the printer send home in x and y axis, but when the printer head reach the center to measure the z axis, the BL touch just send deploy and stow repeatedly and not Go down the z axis?
> 
> exactly that

Check wiring to BL-Touch.  Which way do you wire it? Using the z-probe header, or the Z-Stop?

JustMeJohn44, 2021-12-23:
Im having the same issue. I have it hooked up to z stop and has been working fine until I did a firmware upgrade to my tft35e3v3. Then z homing stopped it raises 10mm and then just repleadly deploys and retracts pin but never drops to bed. I have tried updating firmware but cannot complie for some reason because I have a runout sensor.

marc-gist, 2021-12-23:
> 

Error with your z-stop wire.  check that its plugged in and not lose anywhere.  can also check z-stop by going back to z-stop homing and ensure that the board pins have not been damaged/shorted out.  just be ready to turn off machine before bed crashes. I would just use my finger to click the z-stop.  If that works. then the BLTouch died.

marc-gist, 2021-12-23:
from the code marlin code:
```

   * Note: If the trigger signal of your probe is not being recognized, it has been very often
   *       because the BLACK and WHITE wires needed to be swapped. They are not "interchangeable"
   *       like they would be with a real switch. So please check the wiring first.
```

Wonkothsane, 2021-12-24:
I am having the same issue, but I have a CR Touch.  I believe it is essentially the same hardware.  I had understood though, that it was supposedly already wired to be used as a Z-stop.

JustMeJohn44, 2021-12-24:
Thanks I will got through and double check all my connections.  Might be after the holidays before I can respond back but thank you again.

Shadeaflame, 2022-01-03:
I believe I had the same issue, (Ender 3 v2 BL touch) for what I know is that when disabling my stepper moders I would not be able to change my z axis after auto homer/ auto leveling.

Hansd007, 2022-01-03:
downgrade firmware to previous version, problem should be solved

Shadeaflame, 2022-01-03:
After working to solve the issue I found out that by having extra information (update file, and random stuff) on the SD card will cause glitches, first to troubleshoot this is 
Turn on printer 
auto home/level
attempt to change z if issue persists continue
turn off printer
remove SD card
turn on printer 
move z axis if it moves make note of this
auto home/level
if after homeing you can move the z axis go to the SD card and move every file to a safe place or delete it then 
put in SD 
turn on printer 
move z 
auto home/level
if you can move the Z axis you had the same issue I did and as if something doesn't make sense i did not revise this

Hansd007, 2022-01-04:
There is nothing on my SD card but the firmware. Clean format

Eclipsegsxr, 2022-01-08:
I was having the same issues. I now have a proper firmware.bin using Marlin 2.0.9.1 working for ender 3 pro with skr min e3 v2 and tft v3. I have the latest firmware on my tft. and marlin version is 2.0.9.1. BLTouch is connected to the z-probe on the board and endstops still in place and working. I will home x and y axis then goto center of bed and home z. Also has option in the setting screen to turn on and off for a filament sensor if one is installed.
https://github.com/Eclipsegsxr/Skr-mini-e3-v2

IWantAnAccount, 2022-01-09:
> Check wiring to BL-Touch. Which way do you wire it? Using the z-probe header, or the Z-Stop?

I used the Z-Stop
I found a solution by learning how to compile  Marlin and compiling the firmware myself

IWantAnAccount, 2022-01-10:
If someone want it, I can upload the Marlin folder with the configuration and paste it here

shiftmyhurst, 2022-01-12:
> If someone want it, I can upload the Marlin folder with the configuration and paste it here

that would be helpful been fight with mine for a fuw days now

Eclipsegsxr, 2022-01-12:
i now have latest 2.0.9.3 working on my ender 3 pro with skr e3 v2 and tft35 v3

shiftmyhurst, 2022-01-12:
> i now have latest 2.0.9.3 working on my ender 3 pro with skr e3 v2 and tft35 v3    i am running the same but not sure what i need to put on there to get working i tryed the firmware.bin but i think it is the wrong one for the v2 so if ya have a link to the right firmware that would be a great help

Eclipsegsxr, 2022-01-12:
[https://github.com/Eclipsegsxr/Ender-3-pro-BTT-upgraded](url)

shiftmyhurst, 2022-01-12:
> [https://github.com/Eclipsegsxr/Ender-3-pro-BTT-upgraded](url)

ok i download this file and put on memory card put in printer turned on it loaded fuw seconds later 3 beeps and says failed to enable bed  leveling Try to home x and y home fine go to center sticks prob out starts down and i trip bl touch to see if it will stop it just keeps moveing down

IWantAnAccount, 2022-01-16:
> If someone want it, I can upload the Marlin folder with the configuration and paste it here

https://www.dropbox.com/sh/611tibpwzfpf130/AAAJdiTZqRqawR4JAyjdGbJja?dl=0

Uploaded it. 
Config used: Ender 3, BLTouch connected to Z-stop

donpiphans, 2022-11-06:
I have tried to update the firmware to 2.0.8.2 i used [firmware-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-bltouch-for-z-homing.bin) but now my bl touch does not work.

Eclipsegsxr, 2022-11-07:
> I have tried to update the firmware to 2.0.8.2 i used [firmware-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-bltouch-for-z-homing.bin) but now my bl touch does not work.

scroll up a few and download and install my firmware and you should be good to go.