bigtreetech/BIGTREETECH-SKR-mini-E3 #590: SKR mini E3 V3.0 Firmware for the CR-10S

africsnail, 2021-12-06:
Can you please compile a firmware for the CR-10S?

africsnail, 2021-12-07:
Supposedly, this needs to be solved first: 
https://github.com/MarlinFirmware/Marlin/issues/23189

radek8, 2021-12-07:
https://github.com/bigtreetech/Marlin/tree/SKR-mini-E3-V3.0-G0B1/Marlin

jasonpointer, 2021-12-13:
I'm currently making one for Klipper with no BLtouch and dual Z. 

But cos there are 4 fans in total you will need to make your own Y cable for the 2 case fans to work or just wire it directly to the PSU.

africsnail, 2021-12-13:
Well, I already compiled Marlin. Kinda pieced it out of 2 existing configs. From CR-10S E3 V2.0 and Ender 3 E3 V3.0. Hoping it will work. 

Guess the fans will be issue with Marlin too, but the cable split is quite an easy fix for that.

RockstarSi, 2021-12-24:
Did anyone successfully do this?   I have an CR-10 S5 and wanting to fit the SKR MINI E3 V3 with bl touch. Other than that it’s stock

africsnail, 2021-12-24:
> Did anyone successfully do this? I have an CR-10 S5 and wanting to fit the SKR MINI E3 V3 with bl touch. Other than that it’s stock

Yes, I did compile one for myself. 
I hadn't been able to properly test it yet, since I have a non-compatible LCD (with 2 cables) on my printer. I bought the TFT35 v3 E3 and am waiting for it to arrive.

africsnail, 2021-12-24:
I can try to compile it for you. Just give me the print area sizes. Will do it tomorrow

RockstarSi, 2021-12-25:
Guys if you could that would be fantastic. I had a look into it and this baffles me. Only been printing 4 months and loving it but there’s too much to learn with firmware for the minute. Print area is 500mm x 500mm.  What screen did you buy for your S5 ?

africsnail, 2021-12-25:
I bought the BIQU TFT35 V3 E3. I have CR-10S, not the S5 tho. Will have to print some things, since the screen won't fit the stock assembly.

allaboutmikey, 2022-02-12:
@africsnail I was able to get the stock display working using only one cable by soldering pins into the third header (unpopulated out of the box) on my CR10S. The single cable appears to drive it just fine now from my SKR MINI E3 V3.

dbuck75, 2022-03-06:
@africsnail Would it be possible to get that file complied also  for the skr mini e3 v3.0 with a 300x300x300 build volume.  I also have a BLtouch and the tft35 v3 e3 connected to it.  I added a 300xl ender extender to my ender 3 that I had the skr mini e3 v3.0 board installed in it and was hoping to get updated firmware for it’s new build size.  Bigtreetech’s compiled firmware driver on github called “ender3-bltouch-for-z-homing.bin” for the skr mini e3 v3.0 has been working great. I just need a complied .bin file with the volume upped to 300x300x300.  Thanks for the help if you can.  I’ve no clue how to do it but I’m trying to figure it out.

dbuck75, 2022-03-06:
@africsnail Doh.. The build area is actually 300x300x400

africsnail, 2022-03-06:
> @dbuck75 I can give you my config and it will need just a few changes for you to work.

dbuck75, 2022-03-06:
@africsnail That would be cool.  My email is dbuckles@gmail.com  Thanks a lot.

jasonpointer, 2022-03-07:
> @africsnail Doh.. The build area is actually 300x300x400

Will upload my config when I get home but is incomplete as I've not declared the 3 fans yet and I'm running without bl touch as I've never needed one for my 2 cr10s

Peter2511, 2022-03-10:
Wäre es möglich mir zu helfen. Creality cr10s Board BTT skr3v3 , BLtouch und tft35 v3 e3. 2 Z-Achsen , Filament Sensor. 

Konfiguration benötigt
it-ub@web.de
Vielen Dank

jasonpointer, 2022-03-10:
> > @africsnail Doh.. The build area is actually 300x300x400
> 
> Will upload my config when I get home but is incomplete as I've not declared the 3 fans yet and I'm running without bl touch as I've never needed one for my 2 cr10s

This is not finished yet and requires adjustments before use!!!
[Printer.cfg.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8227490/Printer.cfg.txt)

UncaringWorm, 2022-03-24:
@africsnail I'd love to get that firmware too if you don't mind

SmithGates, 2022-03-31:
I have tried couple times compile my own, that is too difficult :( 
I have a Creality CR10 all original 300x300x400, can some help me please  ? :(

jasonpointer, 2022-04-01:
> I have tried couple times compile my own, that is too difficult :( I have a Creality CR10 all original, no bl touch 300x300x400, can some help me please ? :(

Marlin or Klipper?

rischae, 2022-04-05:
Hey Guys, maybe this will help you :)

https://www.th3dstudio.com/hc/downloads/unified-2-firmware/creality/skr-e3-boards/creality-cr-10s-mini-s4-s5-firmware-skr-e3-mini-v3-board/

I tested it today on my S4 so far with E3 V3.0, TFT35 v3 and BL Touch.

Have some problems with BL-Touch but i'm new to it so can't say whether it's me or the firmware ;)

somroj, 2022-05-29:
Hi,
I did compile my own CR-10S firmware from the Ender 3 BT version. I changed the machine name, build size, adjusted BL touch offset. Everything seems to be working fine and the BL touch can probe the bed correctly in the ABL menu. However, under the L corner sub menu (probably for manual bed leveling) the machine still probe the four corners that seemingly comply to Ender 3 bed size of 235 by 235. I looked everywhere in Configuration.h and Configuration_adv.h files but could not find any parameters that should control this!  Anyone of you with any knowledge please help point me to where to make this adjustment will be highly appreciated. Many thanks in advance.

rischae, 2022-05-30:
@somroj
I guess you are using also the bigtreetech display. I think it has it's own firmware (STM32F207). [Here](https://www.3djake.de/biqu/touchscreen-1?sai=9178&gclid=Cj0KCQjw1tGUBhDXARIsAIJx01k3ebmTFxvQkVwFSlq8mCsBVhtsTmyXBkmt614PBAqPKV5g5gQXrBQaAo_iEALw_wcB)
So you have to modify this, i guess. You can find it here ;) https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

somroj, 2022-05-31:
> @somroj
> I guess you are using also the bigtreetech display. I think it has it's own firmware (STM32F207).
> So you have to modify this, i guess. You can find it here ;) https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Thanks very much, wouldn’t imagine screen has firmware, too! Many many thanks.

michaelruiz2004, 2022-07-24:
can anyone please send me a firmware bin for the cr-10s with cr touch, z offset probe wizard, microswiss hot end, dual geared extruder, hot end temp 300, bed 150 and the printer name being "CR-10S", its for the skr mini e3 v3 and ive been at it for almost 2 full days now with no luck, please someone save me

bera111, 2022-08-16:
Gibt es schon eine Firmware für 1.  SKR mini E3 V3.0 für den CR-10S mit  2.  TFT35 v3.     3.  Bl Touch,  4. Wiederherstellung nach Stromausfall, 5. Filamentsensor, 6. Power off nach Druck.   ?????

1. https://biqu.equipment/collections/control-board/products/bigtreetech-skr-mini-e3-v2-0-32-bit-control-board-for-ender-3
2. https://biqu.equipment/products/btt-tft35-e3-v3-0-display-touch-screen-two-working-modes?variant=31955093815394
3. https://de.aliexpress.com/item/1005001768221232.html?spm=a2g0o.order_list.0.0.274b5c5fsGGmzu&gatewayAdapt=glo2deu
4. https://biqu.equipment/collections/expansion-board/products/bigtreetech-mini-ups-v2-0-module-resume-printing-while-power-off
5. https://biqu.equipment/products/bigtreetech-3d-printer-parts-filament-break-detection-module-for-1-75mm-filament-extruder-material-runout-detector-for-impresora-3d?_pos=4&_psq=filament&_ss=e&_v=1.0
6. https://biqu.equipment/collections/expansion-board/products/bigtreetech-reply-v1-2-automatic-shutdown-module-after-printing

RogueCoyote, 2022-09-06:
> I can try to compile it for you. Just give me the print area sizes. Will do it tomorrow

Hello, if you wouldn't mind, it would be great if you could compile the firmware for me. I have a cr10s with bltouch. Thank you so much!

Peter2511, 2022-09-10:
Trank You, area size are 300 x 300 x 400

Best regards 
Peter

misk0, 2022-11-09:
This is my firmware, Marlin 2.1.1. My hardware is SKR MINI E3 V3.0, BLTouch and TFT35. My BLTouch is Antclabs and I connected 5pin connector directly on board, I've not used Z-Stop connector. I've done only one modification for my setup - I inverted FAN1 with FAN2 PIN because I was connected mainboard and hotend fan in these connectors.
I'm also using Microswiss direct extruder but it has nothing to do with firmware because ESteps are calibrated later, manually.

[SKR_Mini_E3_V3.0_BLTocuh_TFT35.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/9974494/SKR_Mini_E3_V3.0_BLTocuh_TFT35.zip)

SlayHall, 2022-11-26:
Hi, i have made a config for both Klipper and Marlin for the original CR-10 , 
My CR-10 is upgraded with the following:
1-SKR mini v3 e3.
2-BBT smart filament runout sensor (you will have to edit the code if you don't have one ).
3-Duel Z axes (the current is set to 1.3A , if you have a single Z motor set it between 0.530-0.680 Amps). 
4-BTT UPS (this doesn't work on Klipper unfortunately, and from my test  it might work with a Class 10 SDcard but i haven't tried that) 
I didn't have time to do long tests but from what i tried everything works as intended .

cappelletto, 2023-05-10:
@SlayHall cool, I've checked your repo. Have you encountered any issue worth mentioning?

Sciski83, 2025-01-30:
Is there anyone that has good working firmware for the cr10s with the skr e3v3 bltouch, all metal hotend, tft3.5?