bigtreetech/BIGTREETECH-SKR-mini-E3 #593: SKR mini E3 V3.0 Marlin problem with CR Touch z homing

Wonkothsane, 2021-12-25:
Board: SKR mini E3 V3.0
Printer: Ender 3 Pro
Z probe: CR-Touch

### Description
I have installed this Firmware: [firmware-ender3-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin)

### Steps to reproduce
1. Install SKR Mini E3 V3.0 in Ender 3 Pro with CR-Touch
2. Install [firmware-ender3-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin)
3. Auto-Home

**Expected behavior**
Z-Homing success.

**Actual behavior**
When the X & Y Homing complete, the print head goes to the middle of the bed, about 10-20 mm off the bed, then the CR-Touch repeatedly deploys and retracts the pin, the Z-axis does not move otherwise.

### Additional Information
If I install the 3.0 Marlin firmware without the Z-homing, it will use the CR-Touch to ABL and the z-stop, but not the CR-Touch probe for z-Homing.

Taomyn, 2021-12-25:
I was having a similar issue with my v3 but using a custom firmware I'm currently porting from my old v1.2 board, though for me it didn't actually deploy the probe, it would just sit there.

You might fix it by either connecting the black and white cables into the 2-pin Z-endstop port, or if you have your own code enable the following:

// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING

I never needed this on the v1.2 as it was splitting the BL-Touch cables in this way as the probe port was only 3 pins, but now with the v3 it seems to be working now and so the full 5-pin probe socket can be used and not split the cables.

Wonkothsane, 2021-12-25:
> if you have your own code enable the following: 
> // Force the use of the probe for Z-axis homing #define USE_PROBE_FOR_Z_HOMING
>
Will give it a shot, thanks.

Wonkothsane, 2022-01-06:
>if you have your own code enable the following:
>// Force the use of the probe for Z-axis homing #define USE_PROBE_FOR_Z_HOMING
>
>Will give it a shot, thanks.

This issue persists.

Taomyn, 2022-01-07:
@Wonkothsane you're welcome to check out my fork to see if anything of it helps, I customises the README.md with extra info on my hardware and code changes I made

https://github.com/Taomyn/Marlin

Wonkothsane, 2022-01-09:
@Taomyn I appreciate your assistance. Thank you.  I took a look at your Marlin branch. It appears similar to the config I bult locally here.   I got frustrated with it yesterday and built a Klipper firmware and flashed that. I got everything working on the Klipper firmware.  CR Touch works fine and the first layer is buttery smooth.

The only change I made to get the CR Touch working was to add a pull-up to the pin it uses for sensor signal (i.e.: sensor_pin: **^**PC14).  

Perhaps I missed a similar setting in the Marlin Firmware?  If I ever switch back, I will look for a place to check this.

closed 2022-01-09 by Wonkothsane

MavrEars, 2022-03-05:
Here is Ender 3 Pro compiled firmware for SKR mini E3 V3 + CR TOUCH/BL TOUCH bugfix 2.0.x https://disk.yandex.ru/d/EaPAVP5BTo2g5w

Thanks this FAQ: https://www.makenprint.uk/3d-printing/3d-firmware-guides/marlin/consolidated-marlin-guide/

karmavibes7, 2022-05-01:
hi all, i search many informations for the ender 5 cr touch firmware with an skr e3 v3. but i find nothing. 
i don't know what to do, some infos i've read make me think that i have a particular setup and don't find anything working for me.
Can someone help me to understand the problem and find an issue?
thanks

jakedoesgithub, 2022-08-03:
Hi it’s not you it’s the recompiled firmware it’s incorrect here are the older btt version that work https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/d5fc619c2a4aadc889c47680af351310f521e850/firmware/V2.0

TheShield167, 2022-08-08:
Is there a working one for V3.0 yet?

tacopimp, 2026-11-04:
> I fixed that problem with a 3D Touch by using the bug fix release of
> marlin.

unsubscribe