bigtreetech/BIGTREETECH-SKR-mini-E3 #597: [BUG] SD-Card doesn´t rename firmware.bin to FIRMWARE.CUR

rainerschulte, 2021-12-30:
### Description
I have a fresh E3 Mini V3 connected to a Raspberry Pi Zero2 via UART2.
After powering up I expected, that the Filename on the SD-Card (SAN-DISK 32GB, new) would change from **_firmware.bin_** to **_FIRMWARE.CUR_**

### Steps to reproduce

1. I have edited _/boot/config.txt_ and so on and _make menuconfig_ as recomended in the github setup for the Board. 
2. After compiling I copied the _klipper.bin_ to the SD-Card and renamed it to _firmware.bin_.
3. Klipper and the steps to use the E3 MINI V3 are installed on a Raspberry Zero V2 and the Settings for UART2 are set as described here in GitHub. 

**Expected behavior**
Klipper and Mainsail should show the Board up and running.

**Actual behavior**
Klipper has no connection to MCU, 
My Setting in printer.cfg is as follows
`[mcu]
#serial: /dev/serial/by-id/usb-Klipper_stm32g0b1xx_4D003C000B50345033313820-if00
baud: 250000
serial: /dev/ttyAMA0
restart_method: command`

When reading the SD-Card from the SKR Mini V3, the filename is still firmware.bin.

Any Idea what's wrong there?

closed 2022-03-23 by rainerschulte

lukewakeford, 2022-03-24:
@rainerschulte did you resolve this?

rainerschulte, 2022-03-24:
I resolved this because it was working for what reason ever..

The other Issue I have with this board is addressed to Bigtreetech and still not answered.

I decided to use another Board for my Printer.

 

-- 
Der Wahnsinn hat einen Namen!

Rainer (Wahnsinn) Schulte

 
 

Gesendet: Donnerstag, 24. März 2022 um 10:37 Uhr