bigtreetech/BIGTREETECH-SKR-mini-E3 #598: [BUG] SKR Mini E3 V3.0 reboot when trying to heat bed

RodrigoAEscobar, 2021-12-30:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2021-12-31:
I would estimate your problem with a faulty source (dry electrolytic capacitors), or a short circuit in the bed heating heater.
The bed has a large current (10-20A). When you turn on the heating, a poor source may have a power problem.

yansenapdwyll, 2022-02-17:
Well same as above. After switching to old 8bit board works perfectly fine... even bought new DC, nothing works...

TheWhiteDragonsRage, 2023-10-04:
exactly the same issue - seems to be a production fault

ushanovsn, 2024-05-23:
> Well same as above. After switching to old 8bit board works perfectly fine... even bought new DC, nothing works...

I had a similar problem on the SKR MINI E3 V2.0 board. It sometimes rebooted for unknown reason. And once started rebooting whenever the nozzle or bad heating was turned on.
I updated the board firmware, changed the printed wiring, changed the power supply. When everything got really bad, I took off the board and noticed that the main connector 129-5.08-2P (power supply at the input) was slightly loose. The connection was either badly soldered or there was corrosion and the main power contact was not good and became worse over time.
I recommend carefully checking the power connectors and soldering them one more time.