bigtreetech/BIGTREETECH-SKR-mini-E3 #599: [BUG] (Random M1 Stop Commands)

cht685, 2022-01-01:
### Description

After upgrading to the new SKR Mini e3 V3.0 from the V2.0 board, I now receive an error popup for "M1 Stop" which pauses the print until someone clicks the "Continue" button on the screen and/or in Octoprint.

### Steps to reproduce

1. Start Print
2. Wait
3. Click Continue when it pops up
4. Print continues till completion

**Expected behavior**
The print to complete fully without any pop ups that require human intervention.

**Actual behavior**
At some point during the print, usually halfway through, it pops up a "M1 Stop" which stays until someone clicks on the "Continue" button on the TFT35 screen and/or in Octoprint.

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* 
[Configuration.h.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/7798209/Configuration.h.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
* 
[image]

My current setup is as follows:
Ender 3 Pro converted with EnderExtender 400xl kit
Marlin Bugfix 2.0.x downloaded from BTT Github page for the new V3.0 board
SKR Mini e3 V3.0
TFT35 e3 v3.0 connected with both cables to run Marlin and Touchscreen modes
Raspberry Pi 4 running latest version of Octoprint connected with provided USB cable from the SKR Board
Included my config.h and a photo of the message that pops up on the screen.

radek8, 2022-01-01:
the same problem is solved here

cht685, 2022-01-02:
Interesting....ok

closed 2022-01-02 by cht685

Guilouz, 2022-01-02:
> the same problem is solved here
> 
> [bigtreetech/BIGTREETECH-TouchScreenFirmware#2295](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2295)

It's not solved. Workaround is to disable M1 function in Marlin to avoid print paused.

radek8, 2022-01-16:
@cht685 
Kisslorand may have found a solution to the problem.
Do you want to join the repair testing?

mustang5269, 2022-01-16:
I made the change and will test.