bigtreetech/BIGTREETECH-SKR-mini-E3 #600: skr mini e3 v2 BLtouch 2.0 ender 3 cant find z offset---5 pin connector/keeping zstop plugged in

sethboy123456, 2022-01-02:
so I just wired up the skr mini e3 2.0 and flashed the firmware-bltouch.bin from the github page.  The Bl-touch is plugged into the five pin dedicated port on the SKR.  Everything functions as it should, and the probe will complete its 25 point operation and stop. The issue im having is that when i go to print the nozzle is still way to high.  The stock offset is -1.85 and im using a stock extruder/the bracket that came with the BL touch.  How do I find my true z-offset when I'm not using the bltouch as the z-stop? it wont let me move my z value below 0 and there is no function for the probe to find home in the center of the bed like i see in other videos.  I would set it up and use it as the z-stop but my bltouch came with a one piece 5 pin connect.  Please help -Seth

beshur, 2022-01-05:
Same here with SKR Mini v3

Seelenkind, 2022-01-31:
I've read about the problem before.
If you have connected the 2 wires from the BL Touch to the Z endstop then the firmware must contain the following:
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
//#define USE_PROBE_FOR_Z_HOMING
If you have connected all 5 wires to the probe pin, it must be the other way around:
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING

beshur, 2022-02-01:
My problem was fixed by adding spacers to the BlTouch mount, so the
deployed BlTouch pin was below the nozzle.