bigtreetech/BIGTREETECH-SKR-mini-E3 #601: [BUG] SKR mini E3 V3.0 Dwin (ender 3 v2)

spacecabbie, 2022-01-02:
### Description
Referring to: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/424

Applying the same fix to the new board I get no display. 
Figured to check first before i continue trouble shooting.

### Steps to reproduce
compiled firmware from https://github.com/bigtreetech/Marlin/ with:
#define DWIN_MARLINUI_PORTRAIT 
//#error "DWIN_CREALITY_LCD requires a custom cable, see diagram above this line. Comment out this line to continue."

**Expected behavior**
Had hoped it would work out of the box.

**Actual behavior**
compiles fine and firmware is applied (.bin is renamed to .cur) but red led keeps blinking also afther restart.
Note: I cannot exclude that perhaps my display cable broke or something just want to make sure layout is the same.

### Additional Information
[my_config.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/7799924/my_config.zip)

radek8, 2022-01-03:
If necessary, reconnect the cables to the display.
Find inspiration here:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/issues/40

lanzaralx, 2022-01-07:
does it finally work?

spacecabbie, 2022-01-09:
> does it finally work?

Wel yes and no. 
The display wil work but constantly gets garbled i am still tuning it.

tome9111991, 2022-01-12:
Hello.
For a test you can try this Wiring diagram:
https://github.com/MarlinFirmware/Configurations/issues/535#issuecomment-873010783

And add `#define LCD_SERIAL_PORT 1` in your conig.h

tome9111991, 2022-01-22:
If this merged https://github.com/MarlinFirmware/Marlin/pull/23593
You dont need `#define LCD_SERIAL_PORT 1` in your Configuration.h

Dids, 2022-02-01:
> > does it finally work?
> 
> Wel yes and no. The display wil work but constantly gets garbled i am still tuning it.

What fixed it for you?

I'm on the latest bugfix branch, using the E3 V3.0 board config, set `SERIAL_PORT` to `-1` (along with some other config tweaks) and I've triple checked my wiring, however all I see is the backlight coming on the DWIN display, yet no picture of any kind.

I'm still not sure if I'm doing something wrong on the firmware side of things, or if my wiring is either wrong or my cables are faulty.

ingoknito0815, 2022-02-04:
> > > does it finally work?
> 
> > 
> 
> > Wel yes and no. The display wil work but constantly gets garbled i am still tuning it.
> 
> 
> 
> What fixed it for you?
> 
> 
> 
> I'm on the latest bugfix branch, using the E3 V3.0 board config, set `SERIAL_PORT` to `-1` (along with some other config tweaks) and I've triple checked my wiring, however all I see is the backlight coming on the DWIN display, yet no picture of any kind.
> 
> 
> 
> I'm still not sure if I'm doing something wrong on the firmware side of things, or if my wiring is either wrong or my cables are faulty.
> 
> 
> 
> UPDATE: Figured it out. Display was refusing to flash itself.

I bought the skr mini e3 v3 unknowingly of the display problem.

I tried but i just couldn't get the firmware to compile.
Would you be so kind and tell me exactly what you changed to make it work?

neilvangeffen, 2022-02-27:
Thanks, this solved my issues! Wiring and serial_port -1 was what i needed. Massive head ache for the last 4 hours!

Dids, 2022-02-28:
Let me know if you're seeing artifacting with your DWIN display, especially after longer prints, as I've been trying to debug it for a while now, however it might just be a bug in latest Marlin (bugfix-2.0.x).

neilvangeffen, 2022-03-03:
those settings just got it to compile... dont actually have the screen running. Figuring out why its not homing properly first, then im onto why the screen doesnt work :(

Dids, 2022-03-07:
If the configuration is correct and the display has been flashed with the newer firmware, it has to be the custom cable that's not wired correctly, unfortunately.

Dplayer82, 2022-04-06:
> ### Description
> Referring to: #424
>
> Applying the same fix to the new board I get no display. Figured to check first before i continue trouble shooting.
>
> ### Steps to reproduce
> compiled firmware from [...]

Can u send for me firmware? I have same problem (the screen only dark...), but i can't modify marlin...
mrtcsati@gmail.com

nitrospaz, 2022-09-25:
For anyone else that's new to this coming here from google looking how to solve the LCD beep of death problem, before doing any of this for the LCD identify what LCD screen you have first. There are apparently 3 different types of this same display for the Ender 3 V2 that all look identical on the outside. One of the displays can be flashed, another can be flashed but still has errors and the third one apparently has locked down firmware that can't be flashed. Take a guess as to which one I have and how much time I wasted trying to figure this all out... You need to take off the back cover of the screen to identify which one you have. I found this writeup explaining how to identify the screen with pictures here: https://support.th3dstudio.com/helpcenter/creality-printers-color-lcds/

kamisamacash, 2022-12-25:
Do you have the firmware for ender3 v2 with a skr mini e3 v3? could you share it