bigtreetech/BIGTREETECH-SKR-mini-E3 #602: [BUG] Z-Axes does not move down on z-homing (SKR mini E3 V3.0 Marlin, BL Touch Smart v3.1, Ender3 v1)

hieblmedia, 2022-01-03:
### Description

On Auto-Home (and also via moving menu) the Z-Axis does not move back down anymore without the Z-EndStop.
With the EndStop switch plugged in it does move down.

### Steps to reproduce

1. Replaced the old original board with the new SKR mini E3 V3.0
2. Installed the BL Touch Smart v3.1
3. Removed the original Z-EndStop switch.
4. Installed the Firmware: [firmware-ender3-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin)
5. Press Auto-Home

**Expected behavior**
Z-Axis should move down also without the EndStop with [firmware-ender3-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin).

**Actual behavior**
Z-Axies does not move down on Auto-Home or even over the moving menu.
For example set moving +10mm the Z-Axis is moving up, then set back to 0 nothing happens without the EntStop plugged in.

Hansd007, 2022-01-03:
install te previous version firmware, problem should be solved

hieblmedia, 2022-01-03:
Thx for the hint. 
Tried all previous precompiled files. 
The only one works is the first version from 2 month ago, so far on an quick test.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/a3076873ab0b34e1e33d44e56bc86c861c7e3432/firmware/V3.0/Marlin/firmware-bltouch-for-z-homing.bin

In the meantime have also tried to compile the firmware myself and had the same issue.
Now, I am wondering - Is it a problem with Marlin or somthing wrong in the config file.

Hansd007, 2022-01-04:
Why is @bigtreetech not responding on this issue. They should already know there is a problem?

leventuri, 2022-01-05:
I had the same problem. After almost frying my E3V3 board, I could solve the problem changing the sensing wires from Z-PROBE slot to the Z-STOP slot. It seams that we have some problem on Z-PROBE slot or PC14 pin need to be changed in some place of the Marlin code.
-Venturi-

mistic100, 2022-01-05:
Unless I am mistaken "firmware-ender3-bltouch.bin:" should be used if the BlTouch sensor pin is connected to Z-PROBE, and "firmware-ender3-bltouch-for-z-homing.bin" should be used if the BlTouch is connected to Z-STOP.

I installed my V3 yesterday with "firmware-ender3-bltouch-for-z-homing.bin" and using Z-STOP, works perfectly.

bmgermann, 2022-01-05:
I have a similar problem. I installed the firmware-ender3bltouch.bin, and installed the BLTouch to the Z-Probe. However, it worked once correctly, creating a mesh for the plate. After that, however, the probe would only move to the first point and not descend to measure the height of the plate. The printer would then report an error with the BLTouch when it deployed and did not make contact with the plate.

Tr0w, 2022-01-16:
I don't have z-axis homing either. I can't use BTT pre-compiled firmware because I have a Hemera extruder installed. I need to be able to compile Marlin for my setup.

LazeMSS, 2022-01-20:
I have the same problems with both V2 and V3 card:
https://github.com/MarlinFirmware/Marlin/issues/23395

LazeMSS, 2022-01-21:
@bigtreetech could you look into this - I will be happy to provide further debuggin - it also seems like linear advance is not working for the SKR e3 mini v3 and neither is pin debugging.

Seelenkind, 2022-02-06:
Try this. Ia have compiled Marlin 2.0.9.3 with BL Touch for me 
https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin/Ender%203%202.0.9.3%20EN%20%26%20BL%20Touch

LazeMSS, 2022-02-06:
> Try this. Ia have compiled Marlin 2.0.9.3 with BL Touch for me https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin/Ender%203%202.0.9.3%20EN%20%26%20BL%20Touch
> 
> Look at file named "settings changed" for i made changes in config.h and adv config.h

And does this work?

JakeTheDawgg, 2022-02-10:
yes bugfix file labeled for ender-3 on display works with bl touch and no zstop. https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3.git

what is different in that config vs BTT config? i definetly liked that UI better.

Favmir, 2022-02-17:
I spent the entire yesterday trying to figure out what was wrong with my new setup, narrowed it down to the firmware not being able to handle zprobe port input — unlike the original Crealty firmware, the zstop port had to be physically plugged in or the printer would freeze at G28 or G29 with M112 error.
Tried both [firmware-ender3-bltouch.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch.bin) and [firmware-ender3-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin), and neither works without a zstop plugged in.

EduMota, 2022-03-03:
I'm having the same issue, do someone knows whats the different between current firmware and that version?
I really wants to make more changes on firmware.
> Thx for the hint. Tried all previous precompiled files. The only one works is the first version from 2 month ago, so far on an quick test. https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/a3076873ab0b34e1e33d44e56bc86c861c7e3432/firmware/V3.0/Marlin/firmware-bltouch-for-z-homing.bin
> 

> In the meantime have also tried to compile the firmware myself and had the same issue. Now, I am wondering - Is it a problem with Marlin or somthing wrong in the config file.

MavrEars, 2022-03-05:
Have the same issue. Tried my own compiled firmware with this FAQ:
https://www.makenprint.uk/3d-printing/3d-firmware-guides/marlin/consolidated-marlin-guide/
But after UPDATE to Marlin bugfix 2.0.x  it do the PROBE an MESH but when starting to print stuck with error "Printer Halted Please RESET"

@bigtreetech can you HELP us with this? PLEASE RELEASE NEW FIRMWARE fo E3 v3 and CR TOUCH/BL TOUCH/ 3D TOUCH for ender 3 PRO

nofate301, 2022-03-09:
@MavrEars Thank you!  Holy crap, I was going nuts about this and finally yours worked.

ruralTom, 2022-03-14:
> Why is @bigtreetech not responding on this issue. They should already know there is a problem?

Because they already have your money. Each of these boards has been plagued with issues that only get resolved by issuing a new version of the board. Stop doing business with them.

creggerj, 2022-04-08:
I just got this working. had the same issue for a long time.

In all my findings I believe its something to do with safe homing and probing not enabled in the firmware. No sources... It tooke me a while before I was able to make anything work . 

My next venture is getting my DWIN display from an ender V2 to work with this.

austinprimo, 2022-04-20:
> Have the same issue. Tried my own compiled firmware with this FAQ: https://www.makenprint.uk/3d-printing/3d-firmware-guides/marlin/consolidated-marlin-guide/ But after UPDATE to Marlin bugfix 2.0.x it do the PROBE an MESH but when starting to print stuck with error "Printer Halted Please RESET" UPDATE!: Managed that problem was in power loss restore It need to be like that: #define POWER_LOSS_RECOVERY #if ENABLED(POWER_LOSS_RECOVERY) #define PLR_ENABLED_DEFAULT false // Power Loss Recovery enabled by default. (Set with 'M413 Sn' & M500) //#define BACKUP_POWER_SUPPLY // Backup power / UPS to move the steppers on power loss //#define POWER_LOSS_ZRAISE 2 // (mm) Z axis raise on resume (on power loss with UPS) //#define POWER_LOSS_PIN 44 // Pin to detect power loss. Set to -1 to disable default pin on boards without module. //#define POWER_LOSS_STATE HIGH // State of pin indicating power loss //#define POWER_LOSS_PULLUP // Set pullup / pulldown as appropriate for your sensor //#define POWER_LOSS_PULLDOWN //#define POWER_LOSS_PURGE_LEN 20 // (mm) Length of filament to purge on resume //#define POWER_LOSS_RETRACT_LEN 10 // (mm) Length of filament to retract on fail. Requires backup power.
> 
> ```
> // Without a POWER_LOSS_PIN the following option helps reduce wear on the SD card,
> // especially with "vase mode" printing. Set too high and vases cannot be continued.
> #define POWER_LOSS_MIN_Z_CHANGE 0.05 // (mm) Minimum Z change before saving power-loss data
> ```
> 
> NOW everything is working! ENDER 3 PRO CR TOUCH (OR BL TOUCH) + SKR mini E3 v3 + Marlin bugfix 2.0.x Here is the link: [ENDER 3 PRO BTT SKR MINI E3 V3.0 CRTOUCH MARLIN 2.0.x.bin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8191531/ENDER.3.PRO.BTT.SKR.MINI.E3.V3.0.CRTOUCH.MARLIN.2.0.x.bin.zip)

new to 3d printing
does this have filament runout sensor enabled?

morencyam, 2022-07-12:
> Have the same issue. Tried my own compiled firmware with this FAQ: https://www.makenprint.uk/3d-printing/3d-firmware-guides/marlin/consolidated-marlin-guide/ But after UPDATE to Marlin bugfix 2.0.x [...]

Your firmware solved my issue as well. Is there any chance you have the associated config files as well? There are a few other changed I'd like to make as well.

rvsuper8, 2022-08-29:
Yes config files please.

jrgleason, 2022-08-29:
I can confirm that the image worked for me when the others didn't as well

shawnpetros, 2022-09-20:
> Have the same issue. Tried my own compiled firmware with this FAQ: https://www.makenprint.uk/3d-printing/3d-firmware-guides/marlin/consolidated-marlin-guide/ But after UPDATE to Marlin bugfix 2.0.x [...]

so...just installed this firmware, but octoprint is not connecting to my printer. Is there a jumper I need to add or does the firmware need to be updated to allow serial over usb? Kind of lost here since all I wanted to do was add auto bed leveling to my E3P, and it required a mainboard + firmware compiling and it's been sort of a nightmare so far.

dartarien01, 2022-10-08:
> Have the same issue. Tried my own compiled firmware with this FAQ: https://www.makenprint.uk/3d-printing/3d-firmware-guides/marlin/consolidated-marlin-guide/ But after UPDATE to Marlin bugfix 2.0.x [...]

Thank you so much!! I spent hours looking for a fix, probably clicked on at least 40 videos and subreddits etc, your firmware fixed my issue immediately! I made an account just so I could thank you and let others know it worked!!!

LaddiusMaximus, 2022-10-10:
Ive tried all the fixes here but I still cant home my z axis or use my bl touch. Has anyone made any progress on this?

nofate301, 2022-10-10:
Double check everything.

Make sure you're using the right board, the right cr touch. Check all your 
connections.

Test your axis and make sure the right axis moves when you test.

Check your plugs are properly oriented.

Seriously, it sounds like something is either catastrophically wrong with 
the board or a step is being missed.

Are you sure you're flashing the firmware correctly? You've renamed the 
file to firmware.bin and it's at the root of the card? Your memory card is 
good? Have you tested any other memory cards? Remember the card can't be 
too big and it's formatted correctly?

LaddiusMaximus, 2022-10-13:
So I went back and checked everything. It turns out I did not follow the pinout as closely as I thought I had. My zstop pins were reversed which after correcting, everything is now working as it should. Thanks for the heads up. I now have to make sure my filament sensor is working so you may hear from me again. lol

Meatyhog, 2022-11-16:
I'm very new to 3d printing. I've spent hours trying to get my skr mini e3 to work with my ender 3. I am having the same issues as above. It looks like the problem has been solved for the Ender 3 Pro. Will that software work for the standard ender 3?

LaddiusMaximus, 2022-11-21:
I think so? But aurora tech had the fix here: https://youtu.be/dtwWnz2-5VQ

basmul, 2022-11-30:
> Have the same issue. Tried my own compiled firmware with this FAQ: https://www.makenprint.uk/3d-printing/3d-firmware-guides/marlin/consolidated-marlin-guide/ But after UPDATE to Marlin bugfix 2.0.x [...]

Thank you soooo much for your version fixed my issue as well. I also spent three days reading everything, checking everything but to no avail. Now I can finally print some useless things again.

LaddiusMaximus, 2022-11-30:
Not my version, but you are quite welcome. Glad it helped.

Boerny41, 2022-12-28:
Hi,
I've got the same issue but only after upgrading to Marlin 2.1.2. On version 2.1.1 everything worked as intended. I've compared the configuration.h and conf_adv.h from both versions with their counterparts and everything should be the same.

I'd try the version here but I'm not using an Ender3. Did you guys make any notable changes to Marlin but the ones regarding power loss? I've tried these but it doesn't solve the problem. Also like I mentioned it worked on version 2.1.1 with other power loss settings so maybe something else is the reason for it.

Some general information that might be helpful:
Anycubic Mega SE
SKR mini E3 V3
BLTouch Clone (3dTouch I think)
Z endstop not connected to the motherboard
use_zmin_plug not defined in marlin

I've noticed that while USE_ZMIN_PLUG is defined the Z axis does move down when you are inside the move axis menu however it still does not when homing.
While USE_ZMIN_PLUG is not defined it doesn't move in either scenario.

SocalS4, 2022-12-30:
> so...just installed this firmware, but octoprint is not connecting to my printer. Is there a jumper I need to add or does the firmware need to be updated to allow serial over usb? Kind of lost here [...]

Hello, Just wondering if you eve got Octoprint to connect.  This is the only firmware that allows my E3P to auto home correctly.  I have compiled many different versions and have had the serial connection working through the USB to Octoprint but cant get the auto homing to work without using @MavrEars Version.  Does anyone have a decompiles version that is working with Bltouch that i can compare? Or a list of changes needed for the compile to work?   THNAKS

kentonbmax, 2023-01-20:
> I have a similar problem. I installed the firmware-ender3bltouch.bin, and installed the BLTouch to the Z-Probe. However, it worked once correctly, creating a mesh for the plate. After that, however, [...]

How did you get past this? I have same issue.

SocalS4, 2023-01-20:
I ended up having a compile my own firmware from scratch many many times
until I figured out the problems. And there were tons of them. If you'd
like, I can try to find a way to upload the firmware bin file and it will
probably work for you.

Steve Hughes
Numero Uno, Massive WiFi
***@***.*** 1968
S Coast Hwy STE 4883 Laguna Beach, CA 92651

felipecarl, 2023-03-03:
> Have the same issue. Tried my own compiled firmware with this FAQ: https://www.makenprint.uk/3d-printing/3d-firmware-guides/marlin/consolidated-marlin-guide/ But after UPDATE to Marlin bugfix 2.0.x [...]

Thanks!!! With your firmwire now i my printer is working!!

OczO2, 2023-07-14:
> Your firmware solved my issue as well. Is there any chance you have the associated config files as well? There are a few other changed I'd like to make as well.

Unfortunately, after uploading the firmware to the ender 3 V1, I cannot connect the printer to the octoprint, otherwise everything works perfectly, thank you too!
Ps. Do you know how to fix the connection to octoprint.

snador, 2025-01-22:
If you don't use the Z_MIN probe pin, in Marlin 2.x Configuration.h change the following defines:

comment: Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
uncomment: USE_PROBE_FOR_Z_HOMING

With this you disable the Z_MIN input and enable PROBE control for Z homing.
Tested on Ender 3 Pro + SKR mini E3 V3.0 + Marlin 2.1.2.5 with success.