bigtreetech/BIGTREETECH-SKR-mini-E3 #604: [BUG] ( firmware file for BLTouch not working)

Eclipsegsxr, 2022-01-08:
### Description

### Steps to reproduce

1. [Installed BLTouch 3.1 to Ender 3 pro]
2. [Keeping all endstops wired and connecting BLTouch to z-probe port]
3. [Download BLTouch firmware. and copy sdcard and rename.]
4.[put sdcard in board and turn on to install.]
5. [attempt to home all and set z offset. save z offset and probe bed]

**Expected behavior**

**Actual behavior**

### Additional Information
* Given a new version of marlin has come out. I compiled a new setup I have an ender 3 pro. dual z motors. BLTouch 3.1 and all endstops still in place. I have a good and proper working firmware.bin file using Marlin 2.0.9.1 for skr mini e3 v2 with TFT v3 touch screen and BLT hooked to z-probe on the board and using probe to home z axis at the center of the bed,

Seelenkind, 2022-02-02:
Try this one . Fixed bug with BL Touch

[Marlin 2.0.x BugFix SKR Mini E3 V3](https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin/BugFix%20with%20BL%20Touch)

Seelenkind, 2022-02-02:
This one is for E3 V2

/**
 * Enable this option for a probe connected to the Z-MIN pin.
 * The probe replaces the Z-MIN endstop and is used for Z homing.
 * (Automatically enables USE_PROBE_FOR_Z_HOMING.)
 */
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/7987128/firmware.zip)

Eclipsegsxr, 2022-02-02:
Yes, that was what I found to work and get things to follow an expected behavior. I have since moved to the latest marlin release and have it working as well. get a few random oddities that pop up that I'm still sorting out.
the sd card will display not connected at times during long prints, but the prints continue and gives no ill effect.
have an issue using presets to heat the system up and then try to start a print. cause the system to home and then hangup and do nothing. a reset and power cycle clears the issue and using temp adjustments from main home menu works as expected.

krukhlis, 2022-02-03:
Hi there @Eclipsegsxr . 
I'm using E3 v2.0 with BLT pluged to the deticated probe port on the board( instead of Zmin endstop). Everything works like a charm with this repo, you just need to adjust related settings in Configuration.h:
`//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN    //no way, we use dedicated BLT port

// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING

/**
 * Z_MIN_PROBE_PIN
 *
 * Define this pin if the probe is not connected to Z_MIN_PIN.
 * If not defined the default pin for the selected MOTHERBOARD
 * will be used. Most of the time the default is what you want.
 *
 *  - The simplest option is to use a free endstop connector.
 *  - Use 5V for powered (usually inductive) sensors.
 *
 *  - RAMPS 1.3/1.4 boards may use the 5V, GND, and Aux4->D32 pin:
 *    - For simple switches connect...
 *      - normally-closed switches to GND and D32.
 *      - normally-open switches to 5V and D32.
 */
//#define Z_MIN_PROBE_PIN 32 // Pin 32 is the RAMPS default
#define Z_MIN_PROBE_PIN PC14 // Because E3 V2 uses PC14 as a dedicated Probe Pin` 

With these changes everything works. I hope this will help you.
P.S.
Also, please pay attention to proper wiring. Original Antlabs probe used by me has different color coding than cheap chinese clones. Just look at the pinout of the probe socket on E3 V2 and plug corresponding wires to the corresponding pins.

brettvitaz, 2022-02-04:
@Seelenkind Thank you for your firmware file. I'm trying to use your config to compile an English language version, but I can't compile for the V3 board. Did you experience this problem: "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."?

brettvitaz, 2022-02-04:
Your firmware works great, but I require an English language version of the firmware, and it appears that the version you have built is in German. Maybe I'm missing something, but I used your config files with Marlin bugfix-2.0.x branch but I cannot compile. I am getting error messages about libraries not supporting the STM32G board. It appears that you figured out how to compile, so I am curious what I can do to make it work.

Seelenkind, 2022-02-04:
>  but I used your config files with Marlin bugfix-2.0.x branch but I cannot compile. I am getting error messages about libraries not supporting the STM32G board. 

i have crated just a new firmware with Marlin 2.0.9.3 in english language NOT Bugfix. Pls download 2.0.9.3 for edit, not Bugfix

https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin/Ender%203%202.0.9.3%20EN%20&%20BL%20Touch

pls look at settings changed to see what i am changed (commet / uncomment)

https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/Ender%203%202.0.9.3%20EN%20%26%20BL%20Touch/settings%20changed

transport-dot, 2022-02-07:
> Try this one . Fixed bug with BL Touch
> 
> [Marlin 2.0.x BugFix SKR Mini E3 V3](https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin/BugFix%20with%20BL%20Touch)

Thanks for this works great… except the controller fan doesn’t work - i have swapped fans and tried other firmwares to confirm it’s not a broken fan or mosfet on the Skr mini. Any idea what could be causing it? And does your controller fan work?

Seelenkind, 2022-02-07:
This is configured : controller is at pin Fan 2 , the third fan at Board 
0,1,2 

 
 ​#​define​ ​USE_CONTROLLER_FAN 
 ​#​if​ ENABLED(USE_CONTROLLER_FAN) 
 ​  #​define​ ​CONTROLLER_FAN_PIN​ FAN2_PIN       ​//​ Set a custom pin for the controller fan

transport-dot, 2022-02-07:
> This is configured : controller is at pin Fan 2 , the third fan at Board 0,1,2
> 
> ​#​define​ ​USE_CONTROLLER_FAN ​#​if​ ENABLED(USE_CONTROLLER_FAN) ​  #​define​ ​CONTROLLER_FAN_PIN​ FAN2_PIN       ​//​ Set a custom pin for the controller fan

@Seelenkind 
Sorry, I don’t understand what your are trying to explain.
Please could you explain:
Does the controller fan work for you?
Do I need to edit the firmware to get it to work?
Or do I just need to plug the fan into a different socket on the board?

Thanks

Seelenkind, 2022-02-07:
Plug Controller Fan at Fan2 . In Firmware Controller Fan setting is fan2

brettvitaz, 2022-02-07:
@Seelenkind 

Thank you for your response. The English file you provided causes the BLTouch to stop working. I found out why I was unable to compile and all attempts to compile 2.0.9.3 produce the same problem. BLTouch pops in and out several times and homing stops.

transport-dot, 2022-02-07:
> Plug Controller Fan at Fan2 . In Firmware Controller Fan setting is fan2

@Seelenkind 
Yes, I have the controller fan at “fan2”
Part cooling fan is at “fan0”
Hotend fan is at “fan1”.
As I say, the fans work on other versions of the firmware. Yours is the only version that the bl touch works.
Thanks

brettvitaz, 2022-02-07:
@transport-dot You may already be aware, but the controller fan only runs when the stepper drivers are engaged.

transport-dot, 2022-02-07:
> @transport-dot You may already be aware, but the controller fan only runs when the stepper drivers are engaged.

@brettvitaz 
Yes I understand that, thanks

Eclipsegsxr, 2022-02-07:
this started as An issue with the skr mini e3 v2 board. HOW someone introduced e3 v3 firmware into this mix is beyond me. you can not use v3 firmware on a v2 board.

transport-dot, 2022-02-07:
> this started as An issue with the skr mini e3 v2 board. HOW someone introduced e3 v3 firmware into this mix is beyond me. you can not use v3 firmware on a v2 board.

@Eclipsegsxr 
Seelenkind was trying to help you. Check the history of the posts.
The bigger issue here is the poor technical support from big tree tech for their products.

Eclipsegsxr, 2022-02-07:
> @Eclipsegsxr Seelenkind was trying to help you. Check the history of the posts. The bigger issue here is the poor technical support from big tree tech for their products.

I had the issue sorted in the first post i made. I was pointing out the issue to BTT and the fact that the firmware they list doesn't work as intended and posted a copy of the latest at the time of the post to an upgraded marlin with everything working for others to use.
As you look through, you'll see that my hyperlink is now gone to my post, and the hyperlink Seelenkind posted was for Version 3, my issue was with version 2 boards. the version issues are what I think is causing confusion and why some have issue trying the firmware,

matonb, 2022-02-26:
Thanks @Seelenkind, this version of your [firmware](https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/blob/d3f532796dff4cc58291bf5f881a7ae3f83e7097/firmware/V3.0/firmware%20Ender%203%202.0.9.3%20BL%20Touch%20at%20Probe.bin) configuration is working for me (SKR MINI E3 v3)

fishlux, 2022-04-02:
[This one](https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/blob/d3f532796dff4cc58291bf5f881a7ae3f83e7097/firmware/V3.0/firmware%20Ender%203%202.0.9.3%20BL%20Touch%20at%20Probe.bin) Works for me on an E3 w/ BL Touch & SKR Mini E3 V3.  
Thank you so much for this.

tbgoblin, 2022-05-11:
> Try this one . Fixed bug with BL Touch
> 
> [Marlin 2.0.x BugFix SKR Mini E3 V3](https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin/BugFix%20with%20BL%20Touch)

That link is no more working..actually i am facing the same issu with the v3 board.. any suggestion?

J4nRyb1n, 2022-06-25:
For someone, who need help, just like me...

I just tried join bltouch to z-stop pins and in was worked with this firmware normally...
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin

brianvoo, 2023-07-14:
Took me a while to get mine working. https://github.com/brianvoo/BIGTREETECH-SKR-MINI-E3-V3-BLTouch-ZHoming give this try.