bigtreetech/BIGTREETECH-SKR-mini-E3 #605: [REQ] Publish configuration.h/configuration_adv.h for Ender 5 SKR-mini-E3-V3.0

jnko, 2022-01-10:
Could you kindly publish the configuration.h/configuration_adv.h for Ender 5 which was used to build the binaries for SKR-mini-E3-V3.0?
All you released are the ones for Ender 3

Thanks in advance

Yathani, 2022-01-13:
I have configred my own which works great on my moded Ender 5 Pro so check everything to make sure it fits your requirements.

This is based on Marline 2.0.9.3

https://1drv.ms/u/s!AlS3vC1rryvwgts2vSR0S0eOaKRuiA?e=EQR4eY

Good luck

Digolera, 2022-02-15:
https://youtu.be/ugRHsLWH95A

this .bin works to me, but still want a configuration.h  :(

Seelenkind, 2022-03-03:
pls try this. NO BL Touch. if you need BL Touch give feedback
[Ender 5 Marlin 2.0.9.3 SKR Mini E3 V3 NO BL TOUCH.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/8176413/Ender.5.Marlin.2.0.9.3.SKR.Mini.E3.V3.NO.BL.TOUCH.zip)

rhweese, 2023-02-25:
_ also have the Ender 5 (2019) model, I installed the SKR Mini V3 V2.0. After compiling the firmware from github for this setup and flashing the firmware to the printer, I get no movement out any of the stepper motors. The X,Y,Z or E stepper motors show no sign of movement at all, where they were working fine before. The end stop switches are new, all wiring is new, no BL touch_or other add on's. The firmware from Github for the Ender5 with the SKR Mini V3 V2.0. should work without modifications I would assume. At any rate I made no changes. I have tried trouble shooting this for a week and I'm at a total dead end here, If and one has any ideas as to what might be the hang up I would like to hear from you, I could really use some ones help at this point. I am using this file from github if anyone is interested in taking a look, here [https://github.com/alepee/Marlin-Ender-5]. I have tested all hardware and electrical connections just to be sure, but as stated before they were working fine before this happened, so I thing it has to be a firmware issue. I do not know if the steppers motors have been disabled in this firmware or not and if so how could it work for the Ender 5 which is of a box type configuration anyway, what would be the reason for doing that in the first place.  I really could use some help on this one. I've reach a dead end as where to next. Any ideas would help!!!! [Thanks!!!!

rhweese, 2023-03-02:
Hello JNKO! here is the configuration.h/configuration_adv.h files, This files are for the Ender 5. 
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/10871951/Configuration.zip)

rhweese, 2023-03-02:
HELLO SEELENKIND! Thanks for your help, I did go and find a firmware which was in the BIGTREETECK-SKR-mini-E3-master ,V2.0. I found that it was close but I needed to make many changes for it to work for the Ender 5 even though it said it was for the ender 5, i.e. x,y,z, were reversed, print size was off and so on, and more. But you sent me in the right direction to find the files I needed which was a great help so I wanted to say thank you.