bigtreetech/BIGTREETECH-SKR-mini-E3 #607: [BUG] After Mobo Change Stepper Driver are noisly as the original board

fabiuscom, 2022-01-15:
Hi, i have an Ender 3 and today i've installed the Skr mini e3 v1.2 with the firmware from github page, but the stepper driver are noisy like the original card. Some advice? Thank you

radek8, 2022-01-15:
Make sure you have StealthChop (M569) mode turned on.
  If you have activated Hybrid Threshold (M913), check what threshold speeds you have set.

fabiuscom, 2022-01-15:
> Make sure you have StealthChop (M569) mode turned on. If you have activated Hybrid Threshold (M913), check what threshold speeds you have set. Check the communication of the TMC2209 drivers with the M122 command.

Hi, i don't have touch nothing, i've installed the firmware in this page

radek8, 2022-01-15:
Use a USB connection and use the Pronterface application

radek8, 2022-01-15:
Send orders
M503
M122
save Marlin's answers here

fabiuscom, 2022-01-15:
ok thanks

fabiuscom, 2022-01-15:
Connecting...
Printer is now online.
>>> M503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E93.00
echo:; Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.00 R500.00 T500.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.08
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Mesh Bed Leveling:
echo:  M420 S0 Z0.00
echo:; Material heatup parameters:
echo:  M145 S0 H185 B45 F255
echo:  M145 S1 H240 B110 F255
echo:; PID settings:
echo:  M301 P21.73 I1.54 D76.55
echo:; Stepper driver current:
echo:  M906 X580 Y580 Z580
echo:  M906 T0 E770
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 T0 E
echo:; Linear Advance:
echo:  M900 K0.00
echo:; Filament load/unload lengths:
echo:  M603 L350.00 U400.00
>>> M122
SENDING:M122
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK

radek8, 2022-01-15:
StealthChop is on
Hybrid Threshold is off
Junction deviation has a value in the norm.
The motor current, and the number of steps is typical of the Ender
UART mode is functional.
I see no reason why engines should be noisy.

fabiuscom, 2022-01-15:
Ok, i've find the solution: the pin selector on the board as SPREAD. Now it's very quite