bigtreetech/BIGTREETECH-SKR-mini-E3 #608: Skr mini e3 microusb connector snapped off - options?

nkeilar, 2022-01-20:
i find the microusb port is quite weak. It snapped off one of my Skr mini e3 boards today despite taking so much care with it. I attempted to resolder it, but its not very easy. I did have it connected to a raspberry pi running octoprint. I see there is an io port, can I somehow use pins on the board for usb or the io port to avoid having to resolder this connector? other ideas would be much appreciated. 

In general this connector is too weak, should be changed to usb c or something else. 

Thanks in advance

radek8, 2022-01-21:
I would use an external connector and connect to the motherboard. What's your version? Send a photo of the damage.

lovegu, 2022-03-06:
This just happend to me last night, with V3.0 board.
Since I dont use TFT Lcd, I connected
miniE 3 - raspberry pi
RX2 - Pin 8 (GPIO14 TX)
TX2 - Pin 10 (GPIO15 RX)
GND - Pin 6 (GND)

then on raspberry pi ssh
sudo raspi-config

Interface -> Serial
I can't remember what the setting was but I remember I set the first setting to "No" and second one to "Yes"

then edited and added line bloew to boot/config.txt via ssh
dtoverlay=pi3-miniuart-bt

then added
/dev/ttyAMA0
to Octoprint settings->serial connection->additional serial ports

after that reboot both ender 3 and raspberry pi

I was able to connect ender 3 to octoprint via rpi's GPIO

brianredbeard, 2022-04-22:
Same here.  Inspection under a microscope reveals a manufacturing defect.  Looking at the solder joints (including the two THT suport pins) it clear that the micro usb port did not undergo the proper reflow curve.

[image]

On the positive side, if you have a hot air workstation (or are competent with fine pitch soldering) this should be a trivial fix.  The other takeaway is that moving to USB-C would not solve this.  As you can see all of the ground pins are actually THT and thus should have sufficient hold.  The problem is that the solder did not even flow through to the pads on the back side of the board:

[image]

While you're at it though, probably a good idea to add some electronics safe adhesive (e.g. [LOCTITE 3621](https://www.henkel-adhesives.com/us/en/product/electrically-non-conductive-adhesives/loctite_3621.html)) before re-soldering.

brianredbeard, 2022-04-23:
Replacement came yesterday, I noticed that it's date code (labeled `D/C: ` on the box) was 2022.3.18.  My previous board's date code was 2022.2.10.

Unfortunately this one was identical in terms of manufacturing.  I preemptively reinforced the four mechanical pads by applying a bit of solder to the back side of the joints to avoid a repeat of the problem.  If you choose to do this make sure that the solder is pulled into the joint via capillary action.

WebSpectrum, 2024-06-08:
Hello - I have same problem micro USB connector snapped off completely, doesn't seem very strong, how easy is it to repair?

jimmiedave, 2025-01-25:
Simply awful. Terrible QC. Nice board in other regards.

I'm fortunate to have a microscope and a rework station, but it's still very tricky, and my connector pulled up traces when it went.