bigtreetech/BIGTREETECH-SKR-mini-E3 #609: [BUG] Skr mini e3 v1.2 - Cannot setup Klipper to use UART -> Marlin working ok

luisfosoares, 2022-01-22:
### Description

Im trying to setup Klipper to use UART instead of default USB cable
I get no response from the board after flashing the firmware on the board 

Changing back to Marlin it works ok. Same wiring same octoprint different reaction 

### Steps to reproduce

1. Edit klipper config to use serial "USART PA3/PA2" - Using TFT Slot for wiring of tx2/rx2
2. Go in Octoprint and try connection - ttyAMA0 

**Expected behavior**
Connection and response from board

**Actual behavior**
Timout and no connection to board

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
[image]

luisfosoares, 2022-01-24:
Solved. Issue was related with octoprint and not with board. Thanks

knipknap, 2022-05-29:
What was the problem in Octoprint? I am facing a similar issue.