bigtreetech/BIGTREETECH-SKR-mini-E3 #610: [BUG] SKR 2.0 restart every 180 seconds

derbenner, 2022-01-24:
I own a new SKR 2.0 board with F429 chip. I use this Marlon bugfix release but exactly after 180 secods the board is rebooting. Is there a new security feature which needs some input?

I test it blank with one fan, nothing was attached only 24v and a fan. The fan stops after 180 seconds every time. When a display is connected there i can see the reboot.

Compiler is VScode and i set the default_env = BIGTREE_SKR_2_F429

radek8, 2022-01-24:
Wrong category.
Close the ticket and publish here:
https://github.com/bigtreetech/SKR-2/issues

But you already know this address.
https://github.com/bigtreetech/SKR-2/issues/109#issuecomment-1020172117

Randwic, 2022-04-14:
Same problem, Skr 2 reboot every 180s when the sd is in motherboard.