bigtreetech/BIGTREETECH-SKR-mini-E3 #613: BLTouch/extruder will not reach the bed. Pease help!!

brachai, 2022-01-31:
Okay, so I recently installed a BLTouch on my Ender 3 Pro. All of the wiring is correct and all of my firmware is up to date with the proper modifications that are standard for running a BLTouch. For some reason, when I try to test the bed leveling function, it will attempt to probe the bed, but after about 10 ish mm, the bltouch acts like there is nothing below it and will not continue downward towards the bed, but instead will retract and sit. I have put my finger under the BLTouch to see if it would at least read something there, and when it touches my finger, it continues to the next probe point, and will continue to work normally as long as iI touch it with my finger.

All of this to ask, why will the thing not going all the way down to the bed? Is there somewhere in the code that I need to define that I needs to keep going until it touches something or is this some other problem? I am using the SKR mini E3 V2.0 board and the most up to date code published here.

Any help would be greatly appreciated! If i need to take a video for a visual reference I will, just lmk!

radek8, 2022-01-31:
Submit the configuration file to Marlin.

brachai, 2022-01-31:
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/7974776/Marlin.zip)
I attached a zipped version of the "Marlin" folder with the modifications to the code.

Seelenkind, 2022-01-31:
I've read about the problem before.
If you have connected the 2 wires from the BL Touch to the Z endstop then the firmware must contain the following:
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
//#define USE_PROBE_FOR_Z_HOMING
If you have connected all 5 wires to the probe pin, it must be the other way around:
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING
if you downloaded the finished firmware from Github you may have to switch the two wires from one socket to the other.

brachai, 2022-01-31:
I was sifting through the code and found the issue! I forgot to uncomment out one of the needed define codes. Rookie mistake. I appreciate all the feedback! and I will keep in mind the wiring issue in case of future issues.

closed 2022-01-31 by brachai